adafruit/Adafruit-ST7735-Library #25: 1.8" st7735 tft lcd library not work on arduino 1.6.5 ide for nodemcu board

taoloo897, 2015-07-10:
1.8" st7735 tft lcd library not work on arduino 1.6.5 ide for nodemcu board

graphicstest.ino: In function 'void setup()':
graphicstest.ino:71:397: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
In file included from C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX\Adafruit_GFX.cpp:35:0:
C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX\glcdfont.c:8:0: warning: "PROGMEM" redefined [enabled by default]
  #define PROGMEM
 ^
In file included from C:\Users\TAO13\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266/Arduino.h:38:0,
                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX\Adafruit_GFX.h:5,
                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX\Adafruit_GFX.cpp:34:
C:\Users\TAO13\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266/pgmspace.h:15:0: note: this is the location of the previous definition
 #define PROGMEM     ICACHE_RODATA_ATTR
 ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX\Adafruit_GFX.cpp:39:0: warning: "pgm_read_byte" redefined [enabled by default]
  #define pgm_read_byte(addr) (*(const unsigned char *)(addr))
 ^
In file included from C:\Users\TAO13\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266/Arduino.h:38:0,
                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX\Adafruit_GFX.h:5,
                 from C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX\Adafruit_GFX.cpp:34:
C:\Users\TAO13\AppData\Roaming\Arduino15\packages\esp8266\hardware\esp8266\1.6.4-673-g8cd3697\cores\esp8266/pgmspace.h:63:0: note: this is the location of the previous definition
 #define pgm_read_byte(addr)                                              \
 ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp: In constructor 'Adafruit_ST7735::Adafruit_ST7735(int8_t, int8_t, int8_t, int8_t, int8_t)':
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:39:3: error: '_cs' was not declared in this scope
   _cs   = cs;
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:40:3: error: '_rs' was not declared in this scope
   _rs   = rs;
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:41:3: error: '_sid' was not declared in this scope
   _sid  = sid;
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:42:3: error: '_sclk' was not declared in this scope
   _sclk = sclk;
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:43:3: error: '_rst' was not declared in this scope
   _rst  = rst;
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp: In constructor 'Adafruit_ST7735::Adafruit_ST7735(int8_t, int8_t, int8_t)':
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:52:3: error: '_cs' was not declared in this scope
   _cs   = cs;
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:53:3: error: '_rs' was not declared in this scope
   _rs   = rs;
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:54:3: error: '_rst' was not declared in this scope
   _rst  = rst;
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:56:3: error: '_sid' was not declared in this scope
   _sid  = _sclk = 0;
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:56:11: error: '_sclk' was not declared in this scope
   _sid  = _sclk = 0;
           ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp: In member function 'void Adafruit_ST7735::commonInit(const uint8_t*)':
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:322:3: error: 'colstart' was not declared in this scope
   colstart  = rowstart = 0; // May be overridden in init func
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:322:15: error: 'rowstart' was not declared in this scope
   colstart  = rowstart = 0; // May be overridden in init func
               ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:324:11: error: '_rs' was not declared in this scope
   pinMode(_rs, OUTPUT);
           ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:325:11: error: '_cs' was not declared in this scope
   pinMode(_cs, OUTPUT);
           ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:334:3: error: 'cspinmask' was not declared in this scope
   cspinmask = digitalPinToBitMask(_cs);
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:335:3: error: 'rspinmask' was not declared in this scope
   rspinmask = digitalPinToBitMask(_rs);
   ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:350:13: error: '_sclk' was not declared in this scope
     pinMode(_sclk, OUTPUT);
             ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:351:13: error: '_sid' was not declared in this scope
     pinMode(_sid , OUTPUT);
             ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:360:5: error: 'clkpinmask' was not declared in this scope
     clkpinmask  = digitalPinToBitMask(_sclk);
     ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:361:5: error: 'datapinmask' was not declared in this scope
     datapinmask = digitalPinToBitMask(_sid);
     ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:379:7: error: '_rst' was not declared in this scope
   if (_rst) {
       ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp: In member function 'void Adafruit_ST7735::initR(uint8_t)':
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:404:5: error: 'colstart' was not declared in this scope
     colstart = 2;
     ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:405:5: error: 'rowstart' was not declared in this scope
     rowstart = 1;
     ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:409:5: error: 'colstart' was not declared in this scope
     colstart = 2;
     ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:410:5: error: 'rowstart' was not declared in this scope
     rowstart = 3;
     ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp: In member function 'void Adafruit_ST7735::setAddrWindow(uint8_t, uint8_t, uint8_t, uint8_t)':
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:432:16: error: 'colstart' was not declared in this scope
   writedata(x0+colstart);     // XSTART 
                ^
C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:438:16: error: 'rowstart' was not declared in this scope
   writedata(y0+rowstart);     // YSTART
                ^
Error compiling.

framb00s, 2015-12-13:
I have the same problem and found no solution

synthead, 2016-01-13:
I'm seeing the same issue.  I'm using an [Adafruit Feather ESP8266 board](https://www.adafruit.com/products/2821) with Arduino configured via [the recommended instructions](https://learn.adafruit.com/adafruit-feather-huzzah-esp8266/using-arduino-ide).

siliconpit, 2016-02-20:
I have the same exact problem. I am using NodeMCU ver-0.9 and arduino 1.6.5 IDE. 
I have connected two different LCDs - Sainsmart 1.8" with ST7735 and the other one 2.2inch TFT 240x320 but both failed the arduino ide examples "graphicstest". I tried all the different LCD drivers listed in arduino including Adafruit_ST7735, Adafruit_ILI9340, Adafruit_ILI9341. 
Thanks in advance for helping on this problem.

TonvanWissen, 2017-02-17:
Finally I have a working ST7735S, thanks to this library.
I'm using this 1.8" SPI 128x160 TFT display on an Wemos D1 (old version) ESP board.

The lines that made the difference: are lines 178 - 184 in the Adafruit_ST7735.h file"
-----------------------------------------------------------------------
#elif defined(ESP8266)
    volatile uint32_t *dataport, *clkport, *csport, *rsport;
    uint32_t  _cs, _rs, _rst, _sid, _sclk,
    datapinmask, clkpinmask, cspinmask, rspinmask,
    colstart, rowstart; // some displays need this changed

#endif
------------------------------------------------------------------------

There is one small but irritating setback:  In the graphicstest.ino example there occurs a software reset after drawing 4 lines in the testlines() function.  I can't understand why that is ..... 
The other examples all work fine!

Nevertheless: I'm happy with the progress I just made.
Thanks!

ladyada, 2017-05-21:
the current version with latest commits is tested working with ESP8266  :)

closed 2017-05-21 by ladyada