adafruit/Adafruit-ST7735-Library #27: filled circle problem

Fleckz, 2015-11-12:
Hello, there is a problem when using filledCircle().
Here is the test code:
#include <Adafruit_GFX.h>
#include <Adafruit_ST7735.h>
#include <SPI.h>

Adafruit_ST7735 tft = Adafruit_ST7735(10,  8, 9);
  
void setup() {
  Serial.begin(9600);
  tft.initR(INITR_BLACKTAB);
  tft.fillScreen(ST7735_BLACK);
  tft.fillCircle(90, 120, 30, ST7735_RED);
  tft.fillCircle(0, 80, 30, ST7735_RED);
  tft.fillCircle(60, 0, 30, ST7735_RED);
}

void loop() { }

The circle with location x=0,y=80 will be distorted, image here:
http://imgur.com/oyS8L6R

I fixed this by adjusting the following lines in Adafruit_GFX::fillCircleHelper() function, not in this lib though, but GFX lib is a dependency for ST7735 lib.
I changed this:
    if (cornername & 0x2) {
      drawFastVLine(x0-x, y0-y, 2*y+1+delta, color);
      drawFastVLine(x0-y, y0-x, 2*x+1+delta, color);
    }
to this:
    if (cornername & 0x2) {
      if (x0-x >= 0) drawFastVLine(x0-x, y0-y, 2*y+1+delta, color);
      if (x0-y >= 0) drawFastVLine(x0-y, y0-x, 2*x+1+delta, color);
    }

Have not noticed any other problems, so this is a issue and a crappy fix for it! :D

Nan0416, 2018-01-25:
Another way to fix it is through changing the Adafruit_ST7735 drawFastVline.
`if((x >= _width) || (y >= _height) || x < 0) return; // add one more checking`

Fleckz, 2018-01-25:
yep, much more nicer!

PaintYourDragon, 2019-01-30:
Appears this got fixed in the GFX library somewhere along the way.

closed 2019-01-30 by PaintYourDragon