adafruit/Adafruit-ST7735-Library #28: need ESP8266 porting & testing

ladyada, 2016-02-08:


tweedius, 2016-02-24:
I have your ST7735R (1.8") breakout board hooked up to a ESP8266 NodeMCU 12E (4M) and I'm currently trying to diagnose why it doesn't work.  Using the Adruino/ESP8266 libraries.

It doesn't seem to be getting past commonInit().  I added the following code to diagnose:
`// Initialization code common to both 'B' and 'R' type displays
void Adafruit_ST7735::commonInit(const uint8_t *cmdList) {
  colstart  = rowstart = 0; // May be overridden in init func

  pinMode(_rs, OUTPUT);
  pinMode(_cs, OUTPUT);
//
Serial.println("commonInit, pinMode");
//
  csport    = portOutputRegister(digitalPinToPort(_cs));
  rsport    = portOutputRegister(digitalPinToPort(_rs));
//
Serial.println("commonInit, portOutputRegister");
//
  cspinmask = digitalPinToBitMask(_cs);
  rspinmask = digitalPinToBitMask(_rs);
//
Serial.println("commonInit, digitalPinToBitMask");
//

  if(hwSPI) { // Using hardware SPI
#if defined (SPI_HAS_TRANSACTION)
    SPI.begin();
//
Serial.println("commonInit, SPI.begin");
//
    mySPISettings = SPISettings(8000000, MSBFIRST, SPI_MODE0);
//
Serial.println("commonInit, SPISettings");
//`

The error printing to serial is as follows:
`Hello! ST7735 TFT TestcommonInit, pinMode
commonInit, portOutputRegister
commonInit, digitalPinToBitMask
commonInit, SPI.begin
co
Exception (28):
epc1=0x40203114 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffefbb0 end: 3ffefdd0 offset: 01a0

Will continue to see if I can resolve the issue.  Seems to be in the SPI.begin call or the SPISettings call, I can't tell.

Hope this information was useful/what you were looking for.

tweedius, 2016-02-28:
Follow up, this library works with the above listed board.
https://github.com/norm8332/ST7735_ESP8266

It appears to be a fork of your code so obviously big props to you for developing the original code and providing a great breakout.

Thanks!

RanTalbott, 2018-03-31:
Using Arduino 1.8.4, the ESP8266 board package 2.3.0, and library version 1.0.8, I got graphicstest.ino running on a Wemos D1 Mini with a 1.44 inch board with red PCB, yellow pins, and a green tab. I haven't tested it with the 2.4.0 version of ESP8266, because that has a bug that affects Belkin Wemo emulation with Alexa, which is one of my main projects at the moment.

I changed the pin assignments to:

 #define TFT_CS     D0
#define TFT_RST    D2
#define TFT_DC     D1

Instantiated tft with:
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS,  TFT_DC, TFT_RST);

and initialized with:
tft.initR(INITR_144GREENTAB);

Worked fine, except that testlines() runs long enough to trip the watchdog timer, so I threw a couple of delay() calls into it to keep the watchdog from biting.

I got frustrated with having the tests zip by without being able to pause and get a good look at what they were doing, so I modified it to let me select and run the tests individually by sending keystrokes through the serial port. I also added a few other features to it, like printing test timing on the serial console and being able to set the pen colors. I'd be happy to contribute it if you'd like to add it to the examples.