adafruit/Adafruit-ST7735-Library #29: Mirror TFT screen on Y axis direction

Salanova, 2016-02-10:
Hi!
I've just bought a TFT 1.8 screen and I'm trying out ST7735 library and Adafruit GFX and all works very well. The problem is that I need to mirror the screen on Y axis.
I found this addresses searching on internet:

ST7735_cmd(0x36);   // Memory data access control:
						//   MY MX MV ML RGB MH - -
//	ST7735_data(0x00);  //   Normal: Top to Bottom; Left to Right; RGB
//	**ST7735_data(0x80);  //   Y-Mirror: Bottom to top; Left to Right; RGB**
//	ST7735_data(0x40);  //   X-Mirror: Top to Bottom; Right to Left; RGB
//	ST7735_data(0xc0);  //   X-Mirror,Y-Mirror: Bottom to top; Right to left; RGB
//	ST7735_data(0x20);  //   X-Y Exchange: X and Y changed positions
//	ST7735_data(0xA0);  //   X-Y Exchange,Y-Mirror
//	ST7735_data(0x60);  //   X-Y Exchange,X-Mirror
//	ST7735_data(0xE0);  //   X-Y Exchange,X-Mirror,Y-Mirror

What should I modify on the library in order to mirror my screen?

Thank you!!

PaintYourDragon, 2016-02-10:
To avoid future library updates clobbering your work, it might be best to do this in your sketch code by issuing 'raw' commands to the driver. You can see this being done in the 'Uncanny Eyes' project, around line 120 (try different values to get the specific mirroring you're after):
https://github.com/adafruit/Teensy3.1_Eyes/blob/master/uncannyEyes/uncannyEyes.ino

closed 2016-02-10 by PaintYourDragon

spilz87, 2019-04-12:
hello,
I'm trying the same thing, but I Don't find a way to do it in my sketch code.
Changing ST77XX_MADCTL_MY to 0 in Adafruit_ST77xx.h does the mirror I need, but as you said, an update will erase this modification.

I tryed to add in setup() :
  digitalWrite(TFT_CS, LOW);
  digitalWrite(TFT_DC, LOW);
  SPI.transfer(ST77XX_MADCTL); // Older TFT lib
  digitalWrite(TFT_DC, HIGH);
  SPI.transfer(ST77XX_MADCTL_MX | /*ST77XX_MADCTL_MY*/ 0  | ST77XX_MADCTL_RGB);
  digitalWrite(TFT_CS, HIGH);
but it doesn't work.

thanks for your help