adafruit/Adafruit-ST7735-Library #33: 1.44 SPI 128x128 Green Tab

mcauser, 2016-04-20:
I have a Chinese board labelled `1.44' SPI 128*128 V1.1` with a green tab on the protective film.
When using this library, the text and rectangles are drawn off-screen.

To make it work, I modified `void Adafruit_ST7735::initR()` and changed the colstart from `2` to `0` and rowstart from `3` to `32`:

    else if(options == INITR_144GREENTAB) {
      _height = ST7735_TFTHEIGHT_144;
      commandList(Rcmd2green144);
      colstart = 0;
      rowstart = 32;
    }

Does anyone else have a green tab and can confirm this on their board?

[image]

mcauser, 2016-04-20:
#32 would address this

gimpox, 2016-04-24:
Alternatively, you could go for [Sumotoy's libs](https://github.com/sumotoy/TFT_ILI9163C/tree/Pre-Release-1.0r6). Soon the final version.
I've also experienced a screen-coordinate translation by using that 1.1 TFT with yellow pins, but I didn't try your trick, now I'm curious.

larsenglund, 2017-03-17:
I have a TFT with the same IC and SKU but it has a red tab on the screen protector. This TFT "works" with the standard values in the library and gets offset with yours. 
It seems a bit buggy though with white flashes everytime tft.fillScreen(ST7735_BLACK); is called and when the graphicstest enters the loop and starts inverting the display is shifts the whole screen half a screen height down.. kindof.. 
I couldn't get Sumotoys library to work with this display (Sumotoy works fine with a 1.44 with black pcb and green tab)

877dev, 2019-01-30:
@mcauser Hi, I have the exact same display. Did this library get ammended to support 128x128 do you know.

I looked at your above comment, and wasn't sure whether you were altering code in the .h or .cpp files? Thanks

mcauser, 2019-01-30:
@877dev TBH, I haven't used this in ~2 years, so without trawling through the recent commits, I'm not sure where it's at. The alteration was in the .cpp file, and it was to do with the initialisation of the display. There's various displays on the market, each with possible subtle alignment differences and sometimes colloquially referred to by the colour of the tab of the removable protective film.

What's going on is the display manufacturers are combining smaller TFT panels with larger driver chips, so when you init the display driver, you need to clip the driver's display ram with a width, height, row and column offset. The same clipping is used to draw things like lines and rectangles.

There's nothing stopping you from writing to the entire display ram, however, on smaller TFTs, you just won't see some pixels and when you try to do high level operations like lines and graphics, they would be relative to the display ram size, not actual TFT pixels.

877dev, 2019-01-30:
@gimpox thank you!

@mcauser Many thanks for the detailed reply!
I also noticed many different types, mine has the green tag and it does match the 1.44" 128x128 in the Adafruit library. I think I had wiring issues because I tried again today and it works fine :)

Also, I have used Sumotoys TFT_ST7735 library and it works too, for 128x128 you need to edit the 
"TFT_ST7735_settings.h" file as per instructions on his/her page: [updated link](https://github.com/sumotoy/TFT_ST7735/tree/1.0p1)

That makes total sense what you wrote about the larger driver chips, it's more understandable for me that the display ram needs to be clipped. Learn something new every day!

Thanks again!

ladyada, 2019-01-30:
hiya we refactored this library to include
void setColRowStart(int8_t col, int8_t row);
https://github.com/adafruit/Adafruit-ST7735-Library/blob/master/Adafruit_ST77xx.h#L105
however you like!

closed 2019-01-30 by ladyada

gascarcella, 2020-11-02:
> 
> 
>
> It seems a bit buggy though with white flashes everytime tft.fillScreen(ST7735_BLACK); is called and when the graphicstest enters the loop and starts inverting the display is shifts the whole screen half a screen height down.. kindof..

I'm facing this exact same issue with this board. I couldnt get any other library to work with this display, only this one does. But its buggy, making the screen go full white. I realized that if i press and keep pressed the reset button of the arduino, the image will show as expected, but after reset will go full blank again.
Avoid using fillScreen wont change anything