adafruit/Adafruit-ST7735-Library #40: Is there any interest in supportng Deek Robot TFT Esplora compatible module?

davidkwast, 2018-04-06:
Hi. I did some modifications to calibrate the zero position and colors. It mixes the color of the blacktab with colstart/rowstart of the greentab. I am still doing some better tests, because this is my first experience with TFT LCD and Anduino.

If you are interested, I can do a proper pull request.

If it really works without any consequences, I will fork it anyway.

---

void Adafruit_ST7735::initR(uint8_t options) {
...
} else if(options == INITR_DEEKROBOT) {
    commandList(Rcmd2red);
    colstart = 2;
    rowstart = 1;
...
  // if black, change MADCTL color filter
  if ((options == INITR_BLACKTAB) || (options == INITR_MINI160x80) || (options == INITR_DEEKROBOT)) {
    writecommand(ST7735_MADCTL);
    writedata(0xC0);
  }

davidkwast, 2018-04-06:
You can see the test code and some photos of the modulo bellow:

https://github.com/davidkwast/Adafruit-ST7735-Library/blob/master/test.ino
https://github.com/davidkwast/Adafruit-ST7735-Library/blob/master/deekrobot_blacktab.jpg
https://github.com/davidkwast/Adafruit-ST7735-Library/blob/master/deekrobot_fix.jpg
https://github.com/davidkwast/Adafruit-ST7735-Library/blob/master/deekrobot_back.jpg

makermelissa, 2020-02-11:
Hi, thank you. It appears the scope of this is covered by using INIT_GREENTAB and using graphicstest.

closed 2020-02-11 by makermelissa