adafruit/Adafruit-ST7735-Library #43: Zero DMA support

BigCorvus, 2018-05-11:
Hello folks,
The display works fine with the Arduino Zero and this library, however after trying a different library (https://github.com/pantata/TFT9341SPI) it turned out that the latter was much faster usind DMA transfer on the ATSAMD21 chip. I tried incorporating the DMA-relevant functions into the Adafruit library but got stuck. 
Do you think it's possible to add Zero DMA support into your lib?
I think many people would be grateful.
Greetings,
AJ

davidperrenoud, 2018-05-27:
The Gamebuino META is based on an ATSAMD21 and a ST7735, so maybe you can test their code?

Their library is heavily based on the Adafruit one, so it should be possible to merge their changes back:
https://github.com/Gamebuino/Gamebuino-META/tree/master/src/utility/Display-ST7735

ladyada, 2018-05-27:
hand tweaking DMA for each lirbary and each display is not gonna happen :D but what is possible is adding DMA to https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT.cpp the SPI TFT sublibrary, that way ILI9341, ST7735, etc can all benefit. if anyone wants to take it on, we'd be happy to help & merge!

PaintYourDragon, 2019-01-30:
DMA is now an option in the GFX library, for this and other SPI-connected displays. It's not enabled by default (experimental) but needs just a minor change to access. Explained here: 
https://blog.adafruit.com/2018/12/12/faster-color-graphics-adafruit_gfx-gains-dma/

closed 2019-01-30 by PaintYourDragon