adafruit/Adafruit-ST7735-Library #44: Support YELLOWTAB displays

brownjd, 2018-05-20:
For some reason two displays I received with yellow tabs print in mirror mode. After playing with all of the various rotations I found the combination that works. Here are my suggestions. You'll see they don't match any of the existing configurations.

Adafruit_ST7735.h
#define INITR_YELLOWTAB  0x8 //for the goofy backwards displays

Adafruit_ST7735.cpp
Line 199: if ((options == INITR_BLACKTAB) || (options == INITR_MINI160x80) || (options == INITR_YELLOWTAB)) {

Line 219 (Case 0): 
} else if(tabcolor == INITR_YELLOWTAB) {
       writedata(ST77XX_MADCTL_MX | ST77XX_MADCTL_RGB);

Line 239 (Case 1):
} else if(tabcolor == INITR_YELLOWTAB) {
       writedata(ST77XX_MADCTL_MX | MST77XX_ADCTL_MV | MADCTL_MY | MADCTL_RGB);

Line 259 (Case 2):
} else if(tabcolor == INITR_YELLOWTAB) {
         writedata(ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB);

Line 279 (Case 3):
} else if(tabcolor == INITR_YELLOWTAB) {
       writedata(ST77XX_MADCTL_MV | ST77XX_MADCTL_RGB);

makermelissa, 2020-02-11:
Could you give an example of a yellow tab display or possibly submit a PR? I know this was a while ago, but I have yet to see a Yellow Tab display and am not able to verify that this works.

brownjd, 2020-02-11:
Well, they look just like there regular ones except that they display right to left, so like you're looking in a mirror. The only indication was the color on the tab of plastic that you peel off of the screen. I made the fix a while ago in my own project to workaround the problem [here](https://github.com/brownjd/ESPStockTicker/blob/master/ST7735_REV.cpp). Since this was two years ago, the listing on eBay is no longer active but it would look like [this](https://www.ebay.com/itm/1-8-inch-TFT-SPI-LCD-Display-Module-ST7735R-128-160-with-PCB-for-Arduino-/133278702888?hash=item1f08079d28) or like [this](https://www.amazon.com/gp/product/B01KZ7N43M/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1). Sometimes these come with squared off edges for the PCB and sometimes they have rounded edges. I don't remember if this was the differentiating factor.

Martin1161, 2022-02-21:
Hi brownjd. jeah, I got it to work. Thanks for your Github. display. The chinese send me one green TAB (works fine with the original library) and one with the Yellow TAB.
The library Adafruit 7735.ccp had changed in the meantime. However with some tweeking it work fine for my 1.8 TFT SPI 128/160 display (KMR-1.8.SPI). 

No changes in ST7735.h.
Adafruit_ST7735.h
#define INITR_YELLOWTAB 0x8 //for the goofy backwards displays

Things I had to change:
find the right lines,
change writedata (...); into madctl = ... ;

Adafruit_ST7735.cpp
Line 244: if ((options == INITR_BLACKTAB) || (options == INITR_MINI160x80) || (options == INITR_YELLOWTAB)) {

Line 283 was 219 (Case 0):
} else if(tabcolor == INITR_YELLOWTAB) {
madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_RGB;

Line 239 was 239 (Case 1):
} else if(tabcolor == INITR_YELLOWTAB) {
madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MV | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB;

Line 259 was 259 (Case 2):
} else if(tabcolor == INITR_YELLOWTAB) {
madctl = ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB;

Line 279 was 279 (Case 3):
} else if(tabcolor == INITR_YELLOWTAB) {
madctl = ST77XX_MADCTL_MV | ST77XX_MADCTL_RGB;

Martin1161, 2022-02-21:
Line 244 was 199...

ladyada, 2022-02-22:
the tab colors aren't really meaningful if not from adafruit shop. better to submit a PR to allow various configs, or an example of setting registers for custom displays.

brownjd, 2022-02-22:
I think the answer is for the library to allow you to make those adjustments - that is mirror, flip, rotate, etc. Then, let the code calling decide what variant it is. This FR has been out since 2018, so we worked around it.