adafruit/Adafruit-ST7735-Library #47: Version 105 of the Adafruit-ST7735 library breaks the uncanny eyes script

MichaelMeissner, 2018-07-15:
Version 105 of the Adafruit-ST7735 breaks the uncannyEyes sketch.  I was posting about the TFT version of uncanny eyes in:
https://forums.adafruit.com/viewtopic.php?f=8&t=81387

I noticed that there had been new updates recently to the Adafruit-ST7735.  I thought I had grabbed the current version (122) but my version was older (104).  I built the sketch as I mentioned in the forums, and it worked.

Because my version of the uncanny eyes sketch has been heavily hacked, I decided to download everything fresh, and try compiling it with minimal changes (just changing the pins to the pins I'm using, and switching back to the TFT display).  When I did this, and compiled it for Teensy 3.2 using the Teensyduino 1.42 overlay (using the Adafruit version of ST7735, not the PJRC version), I get:

/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_ST7735/Adafruit_ST7735.cpp: In member function 'virtual void Adafruit_ST7735::setRotation(uint8_t)':
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_ST7735/Adafruit_ST7735.cpp:236:6: error: '_xstart' was not declared in this scope
      _xstart = _colstart;
      ^
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_ST7735/Adafruit_ST7735.cpp:237:6: error: '_ystart' was not declared in this scope
      _ystart = _rowstart;
      ^
Error compiling for board Teensy 3.2 / 3.1.

So I used subversion to get older versions of the library.  I discovered that the 104 version of the library built the sketch, but the 105 version does not build it.

ladyada, 2018-07-15:
dont forget to update Adafruit_GFX too, if you do that does it still fail?

MichaelMeissner, 2018-07-15:
I replaced the PJRC version of Adafruit_GFX with the current Adafruit version, and I got a bunch more errors.  I suspect at this point, I'm in a twisty little maze of passages all different:

/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.cpp: In constructor 'Adafruit_SPITFT::Adafruit_SPITFT(uint16_t, uint16_t, int8_t, int8_t, int8_t, int8_t, int8_t, int8_t)':
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.cpp:90:15: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'RwReg* {aka volatile long unsigned int*}' in assignment
     csport    = portOutputRegister(digitalPinToPort(_cs));
               ^
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.cpp:92:15: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'RwReg* {aka volatile long unsigned int*}' in assignment
     dcport    = portOutputRegister(digitalPinToPort(_dc));
               ^
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.cpp:94:17: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'RwReg* {aka volatile long unsigned int*}' in assignment
     clkport     = portOutputRegister(digitalPinToPort(_sclk));
                 ^
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.cpp:96:17: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'RwReg* {aka volatile long unsigned int*}' in assignment
     mosiport    = portOutputRegister(digitalPinToPort(_mosi));
                 ^
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.cpp:99:21: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'RwReg* {aka volatile long unsigned int*}' in assignment
         misoport    = portInputRegister(digitalPinToPort(_miso));
                     ^
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.cpp: In constructor 'Adafruit_SPITFT::Adafruit_SPITFT(uint16_t, uint16_t, int8_t, int8_t, int8_t)':
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.cpp:129:15: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'RwReg* {aka volatile long unsigned int*}' in assignment
     csport    = portOutputRegister(digitalPinToPort(_cs));
               ^
/rock/shared/arduino/teensy-1.42-1.8.5/hardware/teensy/avr/libraries/Adafruit_GFX/Adafruit_SPITFT.cpp:131:15: error: cannot convert 'volatile uint8_t* {aka volatile unsigned char*}' to 'RwReg* {aka volatile long unsigned int*}' in assignment
     dcport    = portOutputRegister(digitalPinToPort(_dc));
               ^
Error compiling for board Teensy 3.2 / 3.1.

So for me personally, I'm going to stay with version 104 of the ST7735 for a bit, since that's the least invasive change, but I suspect it may affect other people.

ladyada, 2018-07-15:
ok we can test with teensy 3.2, that isnt in our travis CI and not sure if every could be given the installation procedure is non-standard

BUT this is all for the best, we cannot support nearly a dozen displays with so many platform variations, so we had to create a superclass called Adafruit_SPITFT which can contain any SPI-based optimizations. ideally PJRC can stuff whatever DMA fun into that, so we dont have multiple '_t3' libraries that slowly drift from the stock library! :)

ladyada, 2018-08-13:
this should b e resolved - reopen if necessary!

closed 2018-08-13 by ladyada