adafruit/Adafruit-ST7735-Library #50: esp8266 st77735 80x160 regression to gfx 1.2.5

tablatronix, 2018-07-27:
esp8266 latest stable

had to roll back gfx lib to 1.2.5

symtoms
blank screen no crash or errors with
graphicstest example.

Looking at diffs...

ladyada, 2018-07-27:
thanx we did a big & much-needed refactor - lemme know if you find something!

tablatronix, 2018-07-27:
https://github.com/adafruit/Adafruit-GFX-Library/pull/169

tablatronix, 2018-07-27:
Not sure if USE_FAST_PINIO was supposed to be working for esp or not, but I do not think it works, and has always been disabled until 1.2.6 via include order

ladyada, 2018-07-27:
ok does it work now?

tablatronix, 2018-07-27:
Yes graphicstest works now on my st7735 on nodemcu v2 with just that change.

tablatronix, 2018-07-27:
I have no other displays atm, I have ssd1306 at home though, I can check it later

ladyada, 2018-08-24:
closing cause it seems this is resolved, reopen if not! :)

closed 2018-08-24 by ladyada