adafruit/Adafruit-ST7735-Library #59: Rotation breaks MADCTL color filter workaround for INITR_BLACKTAB and INITR_MINI160x80

cimba007, 2018-10-03:
- Arduino board:  Pro Mini

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.7

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

```
/***************************************************
  This is a library for the Adafruit 1.8" SPI display.

This library works with the Adafruit 1.8" TFT Breakout w/SD card
  ----> http://www.adafruit.com/products/358
The 1.8" TFT shield
  ----> https://www.adafruit.com/product/802
The 1.44" TFT breakout
  ----> https://www.adafruit.com/product/2088
as well as Adafruit raw 1.8" TFT display
  ----> http://www.adafruit.com/products/618

  Check out the links above for our tutorials and wiring diagrams
  These displays use SPI to communicate, 4 or 5 pins are required to
  interface (RST is optional)
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/

#include <Adafruit_GFX.h>    // Core graphics library
#include <Adafruit_ST7735.h> // Hardware-specific library for ST7735
//#include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
#include <SPI.h>

// For the breakout, you can use any 2 or 3 pins
// These pins will also work for the 1.8" TFT shield
#define TFT_RST  7
#define TFT_CS   9
#define TFT_DC   8

// Option 1 (recommended): must use the hardware SPI pins
// (for UNO thats sclk = 13 and sid = 11) and pin 10 must be
// an output. This is much faster - also required if you want
// to use the microSD card (see the image drawing example)

// For 1.44" and 1.8" TFT with ST7735 use
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS,  TFT_DC, TFT_RST);

// For 1.54" TFT with ST7789
//Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS,  TFT_DC, TFT_RST);

// Option 2: use any pins but a little slower!
//#define TFT_SCLK 13   // set these to be whatever pins you like!
//#define TFT_MOSI 11   // set these to be whatever pins you like!
//Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);

float p = 3.1415926;

void setup(void) {
  Serial.begin(9600);
  Serial.print("Hello! ST77xx TFT Test");

  // Use this initializer (uncomment) if you're using a 0.96" 180x60 TFT
  tft.initR(INITR_MINI160x80);   // initialize a ST7735S chip, mini display

  Serial.println("Initialized");
  tft.setRotation(3);
  uint16_t time = millis();
  tft.fillScreen(ST77XX_BLACK);
  
  tft.setCursor(0,0);
  tft.setTextColor(ST77XX_RED);
  tft.setTextSize(1);
  tft.println("This text is red");

  tft.setTextColor(ST77XX_GREEN);
  tft.setTextSize(1);
  tft.println("This text is green");

  tft.setTextColor(ST77XX_BLUE);
  tft.setTextSize(1);
  tft.println("This text is blue");
}

void loop() {

}
```

Following this simple test sketch with a generic ST7735 (160x80) using the **setRotation** command breaks the workaround for the color filter found in **Adafruit_ST7735.cpp**

**Code snipped:**
```
  // if black, change MADCTL color filter
  if ((options == INITR_BLACKTAB) || (options == INITR_MINI160x80)) {
    startWrite();
    writeCommand(ST77XX_MADCTL);
    spiWrite(0xC0);
    endWrite();
  }
```

See the setRotation function in the same file sets the RGB-Bit to one as opposed in the initialisation snipped above.
```
void Adafruit_ST7735::setRotation(uint8_t m) {
  uint8_t madctl = 0;

  rotation = m % 4; // can't be higher than 3

  switch (rotation) {
   case 0:
     if ((tabcolor == INITR_BLACKTAB) || (tabcolor == INITR_MINI160x80)) {
       madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB;
     } else {
       madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST7735_MADCTL_BGR;
     }
```

I have no idea if RGB and BGR are just mixed up. 

Correcting the code to:
```
   case 3:
     if ((tabcolor == INITR_BLACKTAB) || (tabcolor == INITR_MINI160x80)) {
       madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MV | ST7735_MADCTL_BGR;
     } else {
       madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MV | ST7735_MADCTL_BGR;
     }
```

solves the issues for me.

makermelissa, 2019-05-10:
I was unable to reproduce this issue with the latest version of the libraries (ST7735 and GFX). At some point this was probably fixed. Please feel free to reopen if this continues to be a problem.

closed 2019-05-10 by makermelissa

lydie-g, 2019-07-16:
Hi,
I have similar issues:
My picture is reversed like this:
[image]
It comes from MX, MY and MV parameters, but when I change them in MADCTL  in st7735.c, nothing change and I don't understand why.
Plus colors are reversed (red is cyan, green is magenta and blue is yellow). So I try to use the function nrf_gfx_invert but the green and the blue are still reversed.

InstantArcade, 2019-12-23:
If you call invertDisplay(true) and change your colors from RGB/565 to GBR/565 (e.g. Red will be 0x001F instead of 0xF800) everything will look okay. My offsets were still a bit wrong with rotateDisplay(3) so I modified my Adafruit_ST7735.cpp to this
`...
} else if(options == INITR_MINI160x80) {
    _height   = ST7735_TFTHEIGHT_160;
    _width    = ST7735_TFTWIDTH_80;
    displayInit(Rcmd2green160x80);
    _colstart = 26; // was 24;
    _rowstart = 1; // was 0;
  } else {
...  `

I'd like to be able to change it from my Arduino project so that it doesn't potentially break other things, but the setColRowStart() function is protected.