adafruit/Adafruit-ST7735-Library #60: drawBMP issue with ST7735 library later than 1.1.0

srbonham, 2018-11-05:
- Arduino board:  Arduino Uno

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.7

Using the spitbmp example with the most recent ST7753 library no longer works correctly.  Only the top 20 or so pixels of the bmp go on the screen.  Additionally, if you use rotation, no BMP at all is displayed.  It's like the working canvas of the screen is only the top, even though HEIGHT and WIDTH show 160x128 respectfully.  

Reverting back to 1.1.0 allowed the spitbmp to work and my code works again.

ladyada, 2018-11-05:
what exact hardware are you using

srbonham, 2018-11-05:
I am using this on the Adafruit 1.8 TFT  160x128 breakout with Micro SD

PaintYourDragon, 2018-11-05:
Verify you have both the latest ST7735 and GFX libraries (1.2.4 and 1.3 respectively) installed.
Try the different init flags (INITR_GREENTAB, INITR_REDTAB, INITR_BLACKTAB) and see if any change in behavior.
Tested working on a canonical Arduino Uno and green-tabbed ST7735 breakout with SD (only rotations 0 and 2 work with that example, rotations 1 and 3 are scrambled, but that's a limitation of the spitftbitmap example code and not the library).

PaintYourDragon, 2018-11-05:
If everything else checks out and running latest libraries, try slowing down the SPI rate, near the top of Adafruit_ST77xx.cpp:
`#define SPI_DEFAULT_FREQ         16000000      ///< Default SPI data clock frequency`
Try 8 MHz or even 4 MHz, see if any change.

srbonham, 2018-11-10:
Thanks for your suggestions.

So i went back and verified that I was using the latest libraries.  I went with the latest GFX 1.3.1 and ST7735 1.2.4.   Interesting enough, the parrot.bmp is printing fine with setRotation on 0 and 2, but with 1 and 3 it is scrambled.  My BMP that used to work does not work.  Ive attached that bmp.

I also tried to use INITR_GREENTAB, INITR_REDTACK, and INITR_BLACKTAB respectfully.  No change.

I went through each version, 1.0.0, 1.0.1, 1.0.4, 1.0.6, 1.1.0, and starting 1.2.0 my bmp no longer prints without scrambling.  I changed `#define SPI_DEFAULT_FREQ 16000000 ///< Default SPI data clock frequency` to recommended 8 and 4mhz with no change.  Screen shots attached:

srbonham, 2018-11-10:
[s4black.zip](https://github.com/adafruit/Adafruit-ST7735-Library/files/2568850/s4black.zip)

srbonham, 2018-11-10:
This is example with v1.1.0, Both parrot and s4black bmp files are working correctly, all four rotation constants.

[image]
[image]

srbonham, 2018-11-27:
> Wanted to offer a possible head start on something...it's not quite finished, but we're consolidating all of the BMP-loading examples into a new library. Previously, each screen type had its own BMP-loading example code (sometimes more than one), and over time they'd diverge and have different features and bugs. Mostly bugs.
>
> Will probably be finished this week, but if you'd like to take a peek:
> https://github.com/adafruit/Adafruit_ImageReader
>
> Only one example at the moment -- FeatherWingILI9341.ino -- but hopefully this sufficiently explains the idea, that once you have your TFT object (whether an Adafruit_ST7735 or Adafruit_ILI9341 or a few others, to the library it's all the same), you can pass that as an argument to the reader.drawBMP() function. This should now correctly handle ALL of the ugly cases...clipping, rotation and so forth. Then, with the code all centralized in one place, we'll have a much easier time adding support for other BMP variants in the future.

That sounds great. In the meantime I will try the example code you’ve suggested and report back. Thanks a lot

Shawn

PaintYourDragon, 2019-01-30:
Adafruit_ImageReader library is up and running
https://learn.adafruit.com/adafruit-gfx-graphics-library/loading-images

closed 2019-01-30 by PaintYourDragon