adafruit/Adafruit-ST7735-Library #62: setRowColStart method cannot be used

tsangkaho11, 2018-12-07:
- Arduino board:  Arduino Nano

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.7

- error message: 
graphicstest:78:7: error: 'class Adafruit_ST7735' has no member named 'setRowColStart'

   tft.setRowColStart(32, 2);

       ^

exit status 1
'class Adafruit_ST7735' has no member named 'setRowColStart'

tsangkaho11, 2018-12-07:
The method is defined in class Adafruit_ST77xx which is class that Adafruit_ST7735 derived from. Also, I'm using the latest libraries.

PaintYourDragon, 2018-12-07:
As of 1.2.0 (July 14) there is a setColRowStart() function, this is what's currently in the library.
In the pull request for this the function was called setRowColStart(), but the arguments were switched when it was incorporated.

tsangkaho11, 2018-12-07:
Thanks. That make sense now.

But I got another error:
In file included from -\Arduino\libraries\Adafruit-ST7735-Library-master/Adafruit_ST7735.h:4:0,

                 from -\AppData\Local\Temp\arduino_modified_sketch_854931\graphicstest.ino:26:

-\Arduino\libraries\Adafruit-ST7735-Library-master/Adafruit_ST77xx.h: In function 'void setup()':

-\Arduino\libraries\Adafruit-ST7735-Library-master/Adafruit_ST77xx.h:105:13: error: 'void Adafruit_ST77xx::setColRowStart(int8_t, int8_t)' is protected

     void    setColRowStart(int8_t col, int8_t row);

             ^

graphicstest:78:27: error: within this context

   tft.setColRowStart(2, 32);

                           ^

exit status 1
within this context

The sketch is the one immediately under "Adafruit ST7735 and ST7789 Library"

PaintYourDragon, 2018-12-08:
`             from -\AppData\Local\Temp\arduino_modified_sketch_854931\graphicstest.ino:26:`
"arduino_modified_sketch_854931"

This isn't in the original code.

tsangkaho11, 2018-12-08:
I don't know why but it work fine after I made the method public.
c++ and OOP are really bad