adafruit/Adafruit-ST7735-Library #66: Some screens have non-standard col/row start values

BalintCsala, 2019-04-18:
I received a 1.8" 128x160 LCD from China (ST7735) and tried using the library, but it missed a couple of pixels around the edges (2 rows and 1 column on the top and right sides respectively). 

To solve this issue I made "setColRowStart" public and set the values to 2 and 1.

I'm curious why this function isn't public to begin with. It could solve problems like this very easily.

ladyada, 2019-04-19:
mostly for back compatability, you should make a new type or subclass for your display! :)

Kriss7475, 2019-04-27:
I have found out that [this lcd](https://www.waveshare.com/1.8inch-lcd-module.htm) does have exactly the same issue. Is it okay if I add it to initR and make a PR?

BalintCsala, 2019-04-27:
This may be due to the driver being ST7735**S** not just simply ST7735. Mine has that driver too