adafruit/Adafruit-ST7735-Library #70: Feather M0 with st7735 over sercom SPI

orhanyor, 2019-04-30:
hello, im trying to run tft ST7735 with sercom SPI with no luck so far. i carefully followed the instructions in here https://learn.adafruit.com/using-atsamd ... -a-new-spi and tried implement the required code to a very basic ''hello world'' code to test it.

things to note : if i use regular hw SPI constructor with default MOSI SCK pins screen works perfectly fine. i only move those two pins to sercom SPI pins and use corresponding constructor but i get nothing.
while wired for sercom SPI i changed the constructor to use software SPI again screen works with those pins intended for sercom SPI. like this;
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST); 
along with 
#define TFT_MOSI 11
#define TFT_SCLK 12

so i think my wiring is ok.

if i keep the code above for softwareSPI and also keep the sercom declaration along with pinperipheral for pins 11 and 12 screen still works in software SPI. while all this is happening i get very fast count from 0 to 255 in serial monitor which i believe is a sign at least mySPI is alive.

can anybody please look at the code and see if im doing something wrong. thank you

`#include <Adafruit_GFX.h>    
#include <Adafruit_ST7735.h> 
#include <SPI.h>
#include "wiring_private.h"  //essential for pinperipheral function

#define TFT_CS        6
#define TFT_RST       9 
#define TFT_DC        10

SPIClass mySPI (&sercom1, 13, 12, 11, SPI_PAD_0_SCK_3, SERCOM_RX_PAD_1);  // sercom spi definition

Adafruit_ST7735 tft = Adafruit_ST7735(&mySPI, TFT_CS, TFT_DC, TFT_RST);  // st7735 constructor

void setup() {
  Serial.begin(115200);
  
  mySPI.begin();
  
  pinPeripheral(11, PIO_SERCOM);   //defining pin 11, 12, 13 to be used as sercom pins.
  pinPeripheral(12, PIO_SERCOM);
  pinPeripheral(13, PIO_SERCOM);
  
 tft.initR(INITR_MINI160x80);  

 tft.fillScreen(ST7735_WHITE);  

 tft.setTextWrap(false); 

 tft.setCursor(0, 0); 
 tft.setTextColor(ST7735_BLACK); 
 tft.setTextSize(3); 
 tft.println("Hello");
}
uint8_t i=0;
void loop()     
{
  Serial.println(i);
  mySPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));     // copied this part from adafruit spi example 
  mySPI.transfer(i++);
  mySPI.endTransaction();
}`

orhanyor, 2019-05-02:
i somewhat found where the issue is. original implementation is made by @daniel-stewart which is here https://github.com/daniel-stewart/Adafruit-ST7735-Library 
i installed his version of gfx and st7735 libraries with minor change to constructor and thank god it worked. here is the working code.
so, definitely something is not right with the adafruit version.

````#include <Adafruit_GFX.h>    
#include <Adafruit_ST7735.h> 
#include <SPI.h>
#include "wiring_private.h" 

#define TFT_CS        6
#define TFT_RST       9 
#define TFT_DC        10

SPIClass mySPI (&sercom1, 13, 12, 11, SPI_PAD_0_SCK_3, SERCOM_RX_PAD_1);

Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, &mySPI, TFT_RST);

void setup() {
  Serial.begin(115200);
  
  mySPI.begin();
  
  pinPeripheral(11, PIO_SERCOM);
  pinPeripheral(12, PIO_SERCOM);
  pinPeripheral(13, PIO_SERCOM);
  delay(1000);
  
 tft.initR(INITR_MINI160x80);  

 tft.fillScreen(ST7735_WHITE);  

 tft.setTextWrap(false); 

 tft.setCursor(0, 0); 
 tft.setTextColor(ST7735_BLACK); 
 tft.setTextSize(3); 
 tft.println("Hello");
}
uint8_t i=0;
void loop()  
{
  Serial.println(i);
  mySPI.beginTransaction(SPISettings(8000000, MSBFIRST, SPI_MODE0));
  mySPI.transfer(i++);
  mySPI.endTransaction();
}`````

PaintYourDragon, 2019-05-03:
Fixed in Adafruit_GFX version 1.4.12. It’s still necessary, as you’re doing here, to call a custom SPI SERCOM object’s begin() function and pinPeripheral() on the required pins before tft.initR(). Mentioning it here in case someone comes along with a similar question. Thanks for spotting this! The pass-a-SPIClass thing was originally intended for &SPI1 on the Arduino Due and had never been checked on custom-built SERCOMs like this.

closed 2019-05-03 by PaintYourDragon

orhanyor, 2019-05-03:
having a custom SPI is a great thing if you have multiple spi devices so thanks for adding this feature and keeping it alive :)