adafruit/Adafruit-ST7735-Library #71: Problem With NodeMCU ESP8266 12E

narayanvyas, 2019-05-04:
Hi, the library is not working with my NodeMCU ESP8266 12E v3 CP2102 board. It is working well with Arduino, but in NodeMCU, the board restarts again and again (Watchdog issue). Here is the error log -

% ⸮=⸮⸮⸮Hello! ST77xx TFT Test
Exception (28):
epc1=0x402045c0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x000000c0 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffdc0 end: 3fffffc0 offset: 01a0
3fffff60:  3fffdad0 4023c0c2 3ffee468 3ffee528  
3fffff70:  3fffdad0 00000002 3ffee468 40204ad0  
3fffff80:  3fffdad0 3ffee4d0 3ffee468 40202e38  
3fffff90:  feefeffe feefeffe feefeffe 3ffee528  
3fffffa0:  3fffdad0 00000000 3ffee4f8 40205c0c  
3fffffb0:  feefeffe feefeffe 3ffe84f8 40100a0d

ladyada, 2019-05-05:
try the latest, its probably fixed!

closed 2019-05-05 by ladyada

narayanvyas, 2019-05-05:
> try the latest, its probably fixed!

Thank you very much @ladyada , working well now with the latest library downloaded from GitHub. Please update the library in Arduino Library Manager also so that other users don't face the same problem.

ladyada, 2019-05-05:
it was, it takes a few days for the manager to pick up new libraries