adafruit/Adafruit-ST7735-Library #72: 1.3 ST7789 doesnt work after 1.2.5 release.

orhanyor, 2019-05-09:
Hello, wanted to report an issue with st7789 library. i tried to run it for the first time with my feather m0 and i had a blank screen i thought my display is the issue but downgrading the st7735/89 library to version to 1.2.5 made it work like a charm any other version after that like 1.2.6.. etc gives a blank screen.

ladyada, 2019-05-09:
which display is this?

orhanyor, 2019-05-09:
its the cheapo 240x240 chinese version, may be its not compatible but its interesting that it works up until 1.2.5

ladyada, 2019-05-09:
@makermelissa please check our 240x240 ST7789, see if it still works?

orhanyor, 2019-05-09:
by the way the code i tried is the graphicstest.ino from the library

makermelissa, 2019-05-09:
Ok, thanks. I'll try that one out when I get back.

makermelissa, 2019-05-09:
Thanks for reporting this. I was able to reproduce this.

makermelissa, 2019-05-09:
I think this issue may be related to #64.

orhanyor, 2019-05-10:
yes, seems like the same issue  thank you for your time :)

closed 2019-05-10 by orhanyor