adafruit/Adafruit-ST7735-Library #81: ST7789 setRotation(2) - does not display stuff in proper locations...

KurtE, 2019-06-27:
- Arduino board: Teensy 4 beta (but probably all boards)

- Arduino IDE version 1.8.9 - Teensyduino 1.47- Beta 4 (probably does not apply)

- List the steps to reproduce the problem below 
Using current github code (did not verify on library releases...)  Run the following sketch (edit to your CS, DC RST pins:
```
#include <Adafruit_ST7735.h>
#include <Adafruit_ST7789.h>
#include <Adafruit_ST77xx.h>

#define TFT_CS   10  // CS & DC can use pins 2, 6, 9, 10, 15, 20, 21, 22, 23
#define TFT_DC    9  //  but certain pairs must NOT be used: 2+10, 6+9, 20+23, 21+22
#define TFT_RST   23  // RST can use any pin
#define SD_CS     4  // CS for SD card, can use any pin

Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);

void setup(void) {
  pinMode(SD_CS, INPUT_PULLUP);  // don't touch the SD card
  Serial.begin(9600);
  Serial.print("hello!");

  // Use this initializer if you're using a 1.8" TFT
  //tft.initR(INITR_BLACKTAB);
  // Use this initializer (uncomment) if you're using a 1.44" TFT
  //tft.initR(INITR_144GREENTAB);
  tft.init(240,240);  // use for ILI9488
  
  Serial.println("init");
}

uint8_t rotation = 0;
void loop() {
  tft.setRotation(rotation);
  Serial.printf("Set Rotation: %d width: %d height: %d\n", rotation, tft.width(), tft.height());
  rotation = (rotation + 1) & 0x3;
  elapsedMillis timer;
  uint16_t time = millis();
  tft.fillScreen(ST7735_RED);
  tft.fillRect(tft.width()/2-32, 20, 64, tft.height()-40, ST7735_GREEN);
  tft.fillRect(0, 0, 8, 8, ST7735_BLACK);
  tft.fillRect(tft.width()-8, tft.height()-8, 8, 8, ST7735_WHITE);
  // large block of text
  //delay(2500);
  Serial.println("Hit any key to continue");
  while (!Serial.available()) ; 
  while (Serial.read() != -1) ;
}
```
Should draw the simple screen in rotation 0.  Then enter some line in the Serial monitor, which will now draw in rotation 1, again fine.  Then enter something again to try Rotation 2.  Note: the location of the small black square drawn at 0,0... 

Likewise Rotation 3 is also not right.

Currently debugging like code in our own version (ST7735_t3 -> ST7789_t3), that uses faster SPI output.

KurtE, 2019-06-27:
In the case of the 240x240 display that I am using, I was able to make it work I believe correctly by changing:
```
void Adafruit_ST7789::setRotation(uint8_t m) {

  writecommand(ST77XX_MADCTL);
  rotation = m % 4; // can't be higher than 3
  switch (rotation) {
   case 0:
     writedata(ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB);

     _xstart = _colstart;
     _ystart = _rowstart;
     break;
   case 1:
     writedata(ST77XX_MADCTL_MY | ST77XX_MADCTL_MV | ST77XX_MADCTL_RGB);

     _xstart = _rowstart;
     _ystart = _colstart;
     break;
  case 2:
     writedata(ST77XX_MADCTL_RGB);
 
     _xstart = _colstart;
     _ystart = 0; // _rowstart;
     break;

   case 3:
     writedata(ST77XX_MADCTL_MX | ST77XX_MADCTL_MV | ST77XX_MADCTL_RGB);

     _xstart = 0; //_rowstart;
     _ystart = _colstart;
     break;
  }
}
```
i.e.  the chip can handle 240x320 displays and on these 240x240 you are skipping the first 80 rows of memory... But when you are addressing in the opposite direction the offset of 80 screws it up....

Not sure if there is a cleaner solution, to handle potentially other displays where maybe they address from start of memory or are 240x320

ladyada, 2019-06-27:
yah what we need to do is add a `setRowColStart(x, y)` that you can use to adjust the '0,0' point - would you be interested in adding that as a PR?

KurtE, 2019-06-27:
Maybe - But mostly right now trying to help out on PJRC Teensy 4 (Still beta support).   I have a few updates to some of your libraries to help things run on these boards.  Waiting to see if PaulStoffregen would prefer to issue the PR's... Or if I should just issue them...   Things like:

Things like, need to get Adafruit_GFX updated to build and run on these boards. 
Changes up in my Fork/Branch:  https://github.com/KurtE/Adafruit-GFX-Library/tree/T4_Beta

Also to get them to build run with ILI9341 displays: Not sure if I uploaded changes here yet.

Plus may have changes for this library as well... 

Question is what would setRowColStart(x, y) do?  In particular do you set it once and try to have it work for all 4 rotations.  Or you can  call setRotation and then fix it using this new call? 

Also not sure of testing... That is I don't think I have any of your ST77xx displays....  The ones I have are the 1.3 inch wave share (ST7789) and generic SS7735 ones I got from Amazon/Ebay that were supposed to have some other processor...  And again can not order direct from you (anymore) as don't have mail delivery to our house... So use PMB in town.

ladyada, 2019-06-27:
ok send over the GFX PR's when you're ready. if at some point there can be a CI addition for teensy boards we will run it on all our libraries.

setrowcolstart(x, y) would tell the library the location in chip memory that maps to (0,0) of the TFT - the rotation code would do the right thing. the displays you have are fine to use :)

KurtE, 2019-06-27:
Will do.  

GFX - Will probably go ahead and issue PR.  The only issue left is I disable PROGMEM on the T4 (like the T3.x).  But I detect it is T4 and do an #undef PROGMEM first, as it complains (Warning) that there already is a #define for PROGMEM.  I tried to not do anything but if I don't have it do a #define for PROGMEM it errors...  It is some strange interaction between how Adafruit_GFX.cpp  is including glcdfont.c, as for example if I simply include the  structure: atic const unsigned char font[] PROGMEM = {
in a sketch file and it compiles without problem... 

Will play with the setRowColStart(x, y) function... 

Will test on my displays.  Again tried to order some of your different versions(#358, 3787, 3533), but still can not check out, even with Amazon pay using my default addresses for either Amazon or Paypal)... Wish you would have Grandfathered in accounts that had ordered several times before....  As would have also ordered a RPI4...

ladyada, 2019-06-27:
you can pick up displays from digikey https://www.digikey.com/products/en/optoelectronics/display-modules-lcd-oled-graphic/107?k=adafruit%20358 they ship anywhere!

KurtE, 2019-07-05:
I started playing some with this.  Starting off with our own library st7735_T3 which also does st7789... 

Of trying to add the setRowColStart(x, y) and make it automatically handle the setRotation calls. 

Up till now been doing most of my testing using a couple of: 
https://smile.amazon.com/gp/product/B00V5846YC/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
Which are black and on the back are marked (1.44` SPI 128*128 V2.1)  With these, with our library, to work properly in the 4 orientations,  I have setup (_xstart, _ystart) used in the offsetting:
0:(0, 32), 1:(32,0), 2:(0, 0), 3(0, 0)

A couple of days ago I received a different one:  https://smile.amazon.com/gp/product/B073R6SQRY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1
Red with similar marking, except V1.1

With it to work properly: 
0:(2, 3), 1:(3,2), 2:(2, 1), 3(1, 2)

Right now trying to figure out differences for Rotation 2 and 3 as to handle it.  Obviously the easiest solution is to punt and if the user has a different display they could do: 
```
tft.setRotation(2);
tft.setRowColStart(2, 1);
```
Wondering if maybe this different boards have the ST7735 hardware configured differently where the first one is 128x160 and the second one is 132x162 and as such offset where their actual usable area is?

Thoughts/Suggestions? 

P.S. - I have not yet put in new order to digikey... May soon as building some T4 breakout boards and will probably need some parts that I forgot to order.