adafruit/Adafruit-ST7735-Library #82: Hello,

ManuHallais, 2019-07-10:
Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

makermelissa, 2019-07-10:
Hi, could you be more specific as to the error message? Thanks

ManuHallais, 2019-07-10:
ok so I tried this code : https://github.com/wemos/D1_mini_Examples/blob/master/examples/04.Shields/TFT_1.4_Shield/simple/simple.ino

And there's my error message : 
`C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp: In member function 'void Adafruit_ST7735::initR(uint8_t)':

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:244:40: error: 'sendCommand' was not declared in this scope

     sendCommand(ST77XX_MADCTL, &data, 1);

                                        ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp: In member function 'virtual void Adafruit_ST7735::setRotation(uint8_t)':

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:294:6: error: '_xstart' was not declared in this scope

      _xstart   = _colstart;

      ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:295:6: error: '_ystart' was not declared in this scope

      _ystart   = _rowstart;

      ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7735.cpp:359:40: error: 'sendCommand' was not declared in this scope

   sendCommand(ST77XX_MADCTL, &madctl, 1);

                                        ^

Plusieurs bibliothèque trouvées pour "Adafruit_GFX.h"
Utilisé : C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_GFX
Non utilisé : C:\Program Files (x86)\Arduino\libraries\Adafruit_GFX
Plusieurs bibliothèque trouvées pour "Adafruit_ST7735.h"
Utilisé : C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735
Non utilisé : C:\Program Files (x86)\Arduino\libraries\Adafruit_ST7735
Utilisation de la bibliothèque Adafruit_GFX version 1.2.2 dans le dossier: C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_GFX 
Utilisation de la bibliothèque Adafruit_ST7735 version 1.3.7 dans le dossier: C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735 
Utilisation de la bibliothèque SPI version 1.0 dans le dossier: C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\SPI 
exit status 1
C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST77xx.cpp: In member function 'void Adafruit_ST77xx::displayInit(const uint8_t*)':

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST77xx.cpp:103:35: error: 'sendCommand' was not declared in this scope

     sendCommand(cmd, addr, numArgs);

                                   ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST77xx.cpp: In member function 'virtual void Adafruit_ST77xx::begin(uint32_t)':

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST77xx.cpp:127:3: error: 'invertOnCommand' was not declared in this scope

   invertOnCommand  = ST77XX_INVON;

   ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST77xx.cpp:128:3: error: 'invertOffCommand' was not declared in this scope

   invertOffCommand = ST77XX_INVOFF;

   ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST77xx.cpp:130:24: error: no matching function for call to 'Adafruit_ST77xx::initSPI(uint32_t&, uint8_t&)'

   initSPI(freq, spiMode);

                        ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST77xx.cpp:130:24: note: candidate is:
[... 65 lines cut ...]
   HEIGHT = height;

          ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7789.cpp: In member function 'virtual void Adafruit_ST7789::setRotation(uint8_t)':

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7789.cpp:135:6: error: '_xstart' was not declared in this scope

      _xstart = _colstart;

      ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7789.cpp:136:6: error: '_ystart' was not declared in this scope

      _ystart = _rowstart;

      ^

C:\Users\Emmanuel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Adafruit_ST7735\Adafruit_ST7789.cpp:163:40: error: 'sendCommand' was not declared in this scope

   sendCommand(ST77XX_MADCTL, &madctl, 1);

                                        ^

Erreur de compilation pour la carte LOLIN(WEMOS) D1 mini Pro `

makermelissa, 2019-07-10:
According to the error message, you're only running Adafruit GFX version 1.2.2 and that needs to be updated. The latest version is 1.5.6.

closed 2019-07-10 by makermelissa