adafruit/Adafruit-ST7735-Library #100: Compilation of example is broken

petrkr, 2019-12-07:
there is problem with library, even Example can not be built because missing members in class

Using this constructor and init (just uncommented and commented out in example)
```
Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);
tft.initR(INITR_144GREENTAB); // Init ST7735R chip, green tab
```

```
/tmp/arduino_modified_sketch_873815/graphicstest.ino: In function 'void setup()':
graphicstest:90:7: error: 'class Adafruit_ST7789' has no member named 'initR'
   tft.initR(INITR_144GREENTAB); // Init ST7735R chip, green tab
```

Using example "graphicstest"

Board ESP32 with Arduino-ESP32 SDK 1.0.4

petrkr, 2019-12-07:
If used other init (uncommented other line) There is another error:

```
Compiling library "Adafruit-ST7735-Library"
/home/petrkr/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++ -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/config -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_trace -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/app_update -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/asio -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bootloader_support -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/bt -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/coap -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/console -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/driver -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-tls -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32 -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_adc_cal -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_event -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_client -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_http_server -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_https_ota -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp_ringbuf -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ethernet -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/expat -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fatfs -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freemodbus -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/freertos -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/heap -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/idf_test -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/jsmn -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/json -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/libsodium -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/log -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/lwip -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mbedtls -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mdns -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/micro-ecc -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/mqtt -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/newlib -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nghttp -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/nvs_flash -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/openssl -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protobuf-c -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/protocomm -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/pthread -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/sdmmc -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/smartconfig_ack -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/soc -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spi_flash -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/spiffs -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcp_transport -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/tcpip_adapter -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/ulp -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/vfs -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wear_levelling -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wifi_provisioning -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/wpa_supplicant -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/xtensa-debug-module -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp32-camera -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/esp-face -I/home/petrkr/.arduino15/packages/esp32/hardware/esp32/1.0.4/tools/sdk/include/fb_gfx -std=gnu++11 -Os -g3 -Wpointer-arith -fexceptions -fstack-protector - [... cut ...]
/home/petrkr/Arduino/libraries/Adafruit-ST7735-Library/Adafruit_ST7789.cpp:115:10: error: assignment of read-only member 'Adafruit_GFX::HEIGHT'
   HEIGHT = height;
          ^
/home/petrkr/Arduino/libraries/Adafruit-ST7735-Library/Adafruit_ST7789.cpp: In member function 'virtual void Adafruit_ST7789::setRotation(uint8_t)':
/home/petrkr/Arduino/libraries/Adafruit-ST7735-Library/Adafruit_ST7789.cpp:180:40: error: 'sendCommand' was not declared in this scope
   sendCommand(ST77XX_MADCTL, &madctl, 1);
                                        ^
exit status 1
Error compiling for board ESP32 Dev Module.
```

ladyada, 2019-12-07:
hi maybe you're missing a dependancy or something? travis CI passes fine
https://travis-ci.com/adafruit/Adafruit-ST7735-Library/builds/139870335

closed 2019-12-07 by ladyada