adafruit/Adafruit-ST7735-Library #102: Display randomly shifts with artifacts at the border

xinanhuang, 2019-12-21:
Hi! So I recently connected a st7735s display (80x160) up to my esp32 board and seem to not be able to get rid of random artifacts that cover the border of the screen. It also seems that the display is shifted at times, to which I tried to fix through changing the starting draw location in st77xx.h and changing the size of the draw area. However this doesn’t seem to improve the result, and the display also randomly shifts the displayed content in random direction once in a while.

I’m wondering if this is a timing issue, and if so, how would I be able to fix this? Thanks!

Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

makermelissa, 2019-12-21:
Could you try a slightly older version of the library to see if that fixes the issue? The width and height were recently swapped on this display. Thanks.

xinanhuang, 2019-12-21:
Hi! It seems the issue persisted even I called back to the earlier location.

makermelissa, 2019-12-21:
Ok, good to know I didn't break anything. Thanks for reporting this.

makermelissa, 2020-02-11:
So is this an Adafruit 160x80 display that you are experiencing this with or a different brand?

dsvilko, 2020-04-29:
You can try setting in _ST77xx.h:
`#define ST77XX_MADCTL_RGB 0x08`
and in _ST7735.cpp:
```
_colstart = 26;
_rowstart = 1;
```
Together with the` tft.invertDisplay(true)`, this has fixed the issues I was having with the 0.96" 80x160(RGB)IPS from Aliexpress.