adafruit/Adafruit-ST7735-Library #103: Problem - driving TFT 80x160 ST7735 display

classic-audio, 2019-12-22:
I have now been working on TFT TFT 80x160 ST7735 display for several days until I realised your replay to the previus issue.
It is possible to re-install older versions.
I ha ve 3 major project using this display and after the update to your latest version non of the prjects were funtional.
At the same time I have ordered new displays and of course I was thinking that was the problem.

_But no._
**I replaced the current version of the library with 1.5.5. and all my sketches worked again.**

There is still a color definition problem but I solved that by changing:
File: Adafruit_ST77xx.h
#define ST77xx_MADCTL_RGB 0x00
changed too
#define ST77xx_MADCTL_RGB 0x08

I hope you will fix your "updates"
It will be a long time from now before I will implement updates again !!!

Thank you for opening an issue on an Adafruit Arduino library repository.  To
improve the speed of resolution please review the following guidelines and
common troubleshooting steps below before creating the issue:

- **Do not use GitHub issues for troubleshooting projects and issues.**  Instead use
  the forums at http://forums.adafruit.com to ask questions and troubleshoot why
  something isn't working as expected.  In many cases the problem is a common issue
  that you will more quickly receive help from the forum community.  GitHub issues
  are meant for known defects in the code.  If you don't know if there is a defect
  in the code then start with troubleshooting on the forum first.

- **If following a tutorial or guide be sure you didn't miss a step.** Carefully
  check all of the steps and commands to run have been followed.  Consult the
  forum if you're unsure or have questions about steps in a guide/tutorial.

- **For Arduino projects check these very common issues to ensure they don't apply**:

  - For uploading sketches or communicating with the board make sure you're using
    a **USB data cable** and **not** a **USB charge-only cable**.  It is sometimes
    very hard to tell the difference between a data and charge cable!  Try using the
    cable with other devices or swapping to another cable to confirm it is not
    the problem.

  - **Be sure you are supplying adequate power to the board.**  Check the specs of
    your board and plug in an external power supply.  In many cases just
    plugging a board into your computer is not enough to power it and other
    peripherals.

  - **Double check all soldering joints and connections.**  Flakey connections
    cause many mysterious problems.  See the [guide to excellent soldering](https://learn.adafruit.com/adafruit-guide-excellent-soldering/tools) for examples of good solder joints.

  - **Ensure you are using an official Arduino or Adafruit board.** We can't
    guarantee a clone board will have the same functionality and work as expected
    with this code and don't support them.

If you're sure this issue is a defect in the code and checked the steps above
please fill in the following fields to provide enough troubleshooting information.
You may delete the guideline and text above to just leave the following details:

- Arduino board:  **INSERT ARDUINO BOARD NAME/TYPE HERE**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **INSERT ARDUINO
  VERSION HERE**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

makermelissa, 2019-12-22:
Hi, is the specific Mini TFT display you are using an Adafruit product? Which board/microcontroller are you using to drive your MiniTFT? The reason I ask is because I would like to be able to reproduce this and make sure changing one thing doesn't break things for others. Thanks.

classic-audio, 2019-12-23:
The display (picture attached below.
Microcontroller used:
Arduino UNO 
Arduino Nano
Arduino ProMini3V3

I have been using all these controller because I was trying to indetify whats wrong. UNO and Nano was used with a level shift of course.

I would really hope you also solve the color problem at the same time :)

The problem with the latest of your update is:
no display at all or very weak strange display or the display "snow only".

[image]

makermelissa, 2020-02-11:
It seems there may be a difference between the Adafruit 160x80 and this brand. In regards to the color scheme, I believe the adafruit one is BGR and this one says RGB. There may be other differences.

classic-audio, 2020-02-12:
So no solution at all.

makermelissa, 2020-02-12:
Just no solution yet.

classic-audio, 2020-02-14:
What happens now?

nekuneko, 2020-08-25:
Please check if my pull-request #126 can solve the problem.

marcel-licence, 2021-06-19:
I had also some problems using the display. I created a new define for myself. RGB and BGR was required to change. The offsets were wrong and finally I had to invert the inversion: https://github.com/marcel-licence/Adafruit-ST7735-Library. Seems similar to the pull-request, I didn't check this before trying to fix this on my own. Best regards.