adafruit/Adafruit-ST7735-Library #110: Reversed dimensions 2.0" 320x240 Color IPS TFT

tdestro, 2020-02-07:
I am using
https://www.adafruit.com/product/4311

The example graphicstest does not recognize a portion of the screen, in order to get it to work, the dimensions need reversed inside graphicstest as well as this library in file Adafruit_ST7789.c

```
Adafruit_ST7789::Adafruit_ST7789(int8_t cs, int8_t dc, int8_t rst)
    : Adafruit_ST77xx(240, 320, cs, dc, rst) {}
```

Please fix this so I don't have to maintain a fork, thanks.

On a completely fresh macbook pro I am using arduino ide v 1.8.11 and the newest available libraries.

caternuson, 2020-02-10:
This may be related. Something seems broken with init'ing and/or passing values up the inheritance chain.

Was trying the graphicstest sketch on this:
https://www.adafruit.com/product/4383
which ran, but with some weirdness that appeared to be screen size related.

Came up with this simple test sketch:
```cpp
#include <Adafruit_ST7789.h>

#define TFT_CS         10
#define TFT_RST        9
#define TFT_DC         8

Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);

void setup(void) {
  Serial.begin(9600);
  while (!Serial) {};
  Serial.println(F("Hello! ST77xx TFT Test"));

  tft.init(135, 240);

  Serial.println(F("Initialized"));

  Serial.print("width = "); Serial.println(tft.width());
  Serial.print("height = "); Serial.println(tft.height());
}

void loop() {
}
```

which outputs this on the serial monitor:
```
Hello! ST77xx TFT Test
Initialized
width = 320
height = 240
```

makermelissa, 2020-02-10:
Thanks for reporting this. Gonna take a look at it now.

makermelissa, 2020-02-10:
Ok, I was able to fix it by reversing the numbers in the constructor code. I tested on the 320x240, 240x240, and 240x135 displays with that fix.

@caternuson, the issue you found is due to setRotation changing the width and height to be the raw display width and height. This is because in memory, all of the 7789 displays are 320x240 and we just show a window of them. I believe it was from an older implementation and should probably be updated to work with these displays a bit better.

ladyada, 2020-02-10:
nice thanks!

makermelissa, 2020-02-10:
PR coming soon to a repo near you. ;)

PR #111 (2020-02-10): Fixed reversed display init and it returns the correct size
Fixes #110. This makes it both so it isn't reversed and it now returns the correct size.
closed 2020-02-10 by makermelissa