adafruit/Adafruit-ST7735-Library #116: with custome fonts the background of the text doesn't clear

hamed-ta, 2020-03-21:
by using custome fonts in Adafruit_ST7789 library the background of the text doesn't clear even by using 

  tft.setTextColor(ST77XX_WHITE,ST77XX_BLACK);

  Micro is Atmega 644

[image]
[image]

[image]

PaintYourDragon, 2020-03-22:
This is on purpose and by design. It’s documented in the GFX guide:
https://learn.adafruit.com/adafruit-gfx-graphics-library/using-fonts#using-gfx-fonts-in-arduino-sketches-5-6

closed 2020-03-22 by PaintYourDragon

hamed-ta, 2020-03-24:
so it's better to add the needed code to the library to replace the background of the custom fonts as well as the classic.