adafruit/Adafruit-ST7735-Library #120: 1.5.15 has a regression related to setCursor

Avamander, 2020-06-02:
After updating to 1.5.15 the `setCursor` function now has an offset and text is printed off-screen. Everything works correctly in 1.5.13.

I'd estimate the offset to be approximately 6-7 columns of text (at 8px high font), but it's kinda hard to tell exactly. There seems to be no offset when no rotation is specified, however in order to display text correctly I have to use `setRotation(3)`.

My environment:
I'm using a TTGO T-Display module (ESP32), with an ST7789 display attached.

makermelissa, 2020-06-03:
Hi,  I don't have a TTGO T-Display module and I was unable to reproduce with a Huzzah32 (ESP32) feather and Adafruit 1.14" TFT. I ran graphicstest which makes use of setCursor.

Do you have some example code that demonstrates that it is doing this or is it possible that the issue is specific to the TTGO T-Display module?

makermelissa, 2020-06-03:
I'll keep playing around with doing some sanity checks as this may be related to #119.

Avamander, 2020-06-03:
Code I can reproduce this with:
```
#include <Adafruit_GFX.h>
#include <Adafruit_ST7789.h>
#include <SPI.h>

#define TFT_MOSI 19
#define TFT_SCLK 18
#define TFT_CS 5
#define TFT_DC 16
#define TFT_RST 23
#define TFT_BL 4

Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);

void setup() {
  pinMode(TFT_BL, OUTPUT);
  digitalWrite(TFT_BL, HIGH);

  tft.init(135, 240);
  tft.fillScreen(ST77XX_BLACK);
  tft.setTextSize(1);
  tft.setRotation(3);
  tft.setCursor(0, 0);
  tft.println("Test");
}

void loop() {
  delay(1);
}
```

Avamander, 2020-06-03:
On `1.5.13` I see `Test`, on `1.5.15` I see a blank screen.

makermelissa, 2020-06-03:
Thanks. I think I may be narrowing it down.

Avamander, 2020-06-03:
Looking at the back side of my T-Display's display I see `LH114T-IGO1` `1908015-3x-A` `Li`?`it`?. Maybe that's useful information if you wish to compare the display models.

makermelissa, 2020-06-03:
I was able to reproduce using your code (modified for the Feather Huzzah32). I'm pretty sure this is related to using setRotation(3) and PR #113.

PR #122 (2020-06-03): Fix previous change that broke rotation 2 and 3
Fixes #119. Fixes #120.
closed 2020-06-03 by makermelissa