adafruit/Adafruit-ST7735-Library #123: ST7789 with 240X240 drive failure, but 240*135 succeed?

KiraVerSace, 2020-06-04:
Hi, As I said, when I use the Adafruit ST7789 Library drive my TFT, I have several TFTs.
When I drive the 240x135 it succeed, but I change to the 240X240 and it is failed, So I try another Library in the git
https://github.com/ananevilya/Arduino-ST7789-Library
it succeed, I don't know why and the differences between them?

And I find a very interesting question, When I use your  library I must connect the CS pin to the IO, and control by your Library, even I connect it to the GND, the TFT never display.  And my 240X240 TFT's CS pin was connected to the GND in the PCB, So I can not control the CS pin, May be it's the reason that cause it? May  the problem is not your library but the CS pin can not connect to the GND directly?

Could you help me, Thankyou very much.

closed 2020-06-04 by ladyada

KiraVerSace, 2020-06-05:
> we dont support breakouts without CS - please use that library instead!

Thank you for your help!
And I want to know why, because the CS pin connect to GND means the chip is selected, in general,SPI with one device, we could not control the CS pin.

Can you give me the reason?

ladyada, 2020-06-05:
it is incompatible. you cannot support both. we support CS pin only. use other library please