adafruit/Adafruit-ST7735-Library #125: 1.8 TFT SPI 128x160 very slow with Arduino Nano BLE 33

GitLorenzo, 2020-07-05:
Dear all,

I tested the basic example for the 1.8 TFT SPI 128x160 from Adafruit_ST7735 library (graphic test) with 3 different boards:

- Arduino UNO
- Arduino NANO
- Arduino NANO BLE 33 SENSE

The test code runs perfectly and the refresh time is very fast with the UNO and the NANO, but is incredibly slow with the new NANO BLE 33 SENSE. People more expert than me on Arduino Forum suggested that the speed is slower because the Nano 33 BLE is running mbed OS in the background and the Arduino interface library probably has to negotiate with the OS before the sketch code can use the SPI port and I/O pins.

Is there anything I can do to solve this issue?

One note about the connections:

On both UNO and NANO:
LED > 3V3
SCK > 13
SDA > 11
A0 (DC) > 9
RESET > 8
CS > 10
GND > GND
VCC > 5V

On the new NANO BLE 33 SENSE:
LED > 3V3
SCK > 13
SDA > 11
A0 (DC) > 9
RESET > 8
CS > 10
GND > GND
VCC > 3V3

Thank you!

GitLorenzo, 2020-10-29:
Hello, I was wondering if a solution can be found for this issue...

ladyada, 2020-10-29:
hi we dont own one of these, you may want to ask Arduino or mbed for advice - make sure you use hardware SPI fior any displays, software SPI is slow

closed 2020-10-29 by ladyada