adafruit/Adafruit-ST7735-Library #129: [Issue] 1.44 SPI 128x128 Green Tab Screen White Flash

gascarcella, 2020-11-02:
As metioned in this post https://github.com/adafruit/Adafruit-ST7735-Library/issues/33 

I have a TFT with the same IC and SKU but it has a red tab on the screen protector. This TFT "works" with the standard values in the library and gets offset with yours. 
It seems a bit buggy though with **white flashes everytime tft.fillScreen(ST7735_BLACK); is called** and when the graphicstest enters the loop and starts inverting the display is shifts the whole screen half a screen height down.. kindof.. 
I couldn't get Sumotoys library to work with this display (Sumotoy works fine with a 1.44 with black pcb and green tab)

_Originally posted by @larsenglund in https://github.com/adafruit/Adafruit-ST7735-Library/issues/33#issuecomment-287464827_

I'm facing a similar issue. The screen will flash to full white after a few seconds of use. It will remain like that unless i press the reset button on the Arduino board. 
When pressing the reset button, the white flash disappear and **i can see the screen**, all colors and letters are on place. Keeping the reset button pressed will make the screen keep working without white flashes. After the arduino reboots and the code start, the screen will go flash white again.

- Testing with Board: Arduino UNO [ I have several other MCU in which i could and will test ]
- No arduino ide, i'm using plattformio toolchain on CLion IDE

Note: Sometimes it works for more time, sometimes it work for less time. I could make the screen work with another library that I've found somewhere on elecrow, and it does work fine (so im guessing that there are no hardware problems). I've also tried testing on arduino IDE and facing the same issue

I'm not sure what is causing the problem but if someone can guide me a bit on where to look, i could try to check out the library code and see if i can provide a fix.

microkom, 2021-05-13:
I have the same screen: 1.44' SPI 128x128 V2.0 [Blueish (teal) PCB, Green Tab Screen].

I was having the same problem. I had connected the VCC pin to 3.3V. and after connecting to the 5V pin the white screen is gone. I hope it helps.

leo-smi, 2024-05-04:
> I have the same screen: 1.44' SPI 128x128 V2.0 [Blueish (teal) PCB, Green Tab Screen].
> 
> I was having the same problem. I had connected the VCC pin to 3.3V.
> 
> Solution: connect VCC to the 5V pin
> 
> The white screen is gone. I hope it helps.

didn't work for me