adafruit/Adafruit-ST7735-Library #130: Updating my master branch from the upstream(this one) branch gives me messages...

KurtE, 2020-11-07:
Not sure if I should do anything with this, but I often sync up my branches to the main official released branch in particular, 
I may often do:
```
git fetch upstream
git checkout master
git reset --hard upstream/master
git push origin master --force
```

Such that my master branch is in the same state as the released state.  

I did this earlier this wekk and received a message from github:
[KurtE/Adafruit-ST7735-Library] Run failed: Arduino Library CI - master (55bed63)

[KurtE/Adafruit-ST7735-Library]     Arduino Library CI workflow run |   | [KurtE/Adafruit-ST7735-Library]     Arduino Library CI workflow run |  
-- | -- | -- | --
 
[KurtE/Adafruit-ST7735-Library]     Arduino Library CI workflow run
 

Arduino Library CI / build     (main_platforms)      Failed in 7 minutes and 59 seconds                1 |   |   | Arduino Library CI / build     (main_platforms)      Failed in 7 minutes and 59 seconds | 1
-- | -- | -- | -- | --
  |   | Arduino Library CI / build     (main_platforms)      Failed in 7 minutes and 59 seconds | 1

Again I have not done anything to the library here was just getting my fork into sync with yours.  Not sure if there is anything I should do, and/or if you wish to do something about it at your end.

But feel free to close this one out

ladyada, 2020-11-09:
you can check the Actions CI tab to see what failed. we use it to make sure folks do not submit breaking PRs :)

KurtE, 2020-11-09:
Thanks @ladyada  - As you probably remember I ran into some stuff earlier trying to update code... 
But thought I would mention it as I had not changed any code, but simply was getting my master branch of my fork in sync with the "Upstream" forks master branch... 

Which I would assume that implies that your master branch is failing your checks.

ladyada, 2020-11-09:
looks ok as of about 4 days ago - https://github.com/adafruit/Adafruit-ST7735-Library/actions 
whats the exact error you're getting

KurtE, 2020-11-09:
Probably nothing to be done,  I think it was complaining about library properties…

 

In both builds it complained about:

name=Adafruit ST7735 and ST7789 Library

	

 

version=1.5.18

	

 

version=1.6.0

	

 

author=Adafruit

	

 

maintainer=Adafruit <info@adafruit.com <mailto:info@adafruit.com> >

	

sentence=This is a library for the Adafruit ST7735 and ST7789 SPI displays.