adafruit/Adafruit-ST7735-Library #131: Not able to get working with a ST7789 240x240 and an ESP8266

witnessmenow, 2020-12-04:
Hello,

I have been trying to get this library working with my [D1-Mini-TFT shield](https://github.com/witnessmenow/D1-Mini-TFT-Shield) and so far I've had no luck. The shield is working with a different library, so I presume I must be configuring something wrong.

I'm trying to use the "graphicstest" example and I only get a blank screen.

The shield is connected as follows:

| ST7789     | D1 Mini (Generic GPIO) | 
| ---------- | ------- |
| COPI (SDA on most breakouts) | D7 (13)     |  
| DC         | D6 (12)     |  
| SCLK (SCLon most breakouts) | D5 (14)     |  
| RESET      | D0 (16)     | 
| CS         | GND     | 
| Backlight  | D4 (2)     | 

As I'm using the hardware SPI pins for COPI/MOSI and SCLK, I'm using the "OPTION 1" constructor

`Adafruit_ST7789 tft = Adafruit_ST7789(-1, D6, D0); //-1 as CS is grounded`

-1 for CS seems to be supported in [Adafruit_SPITFT.cpp](https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT.cpp#L440) so I think that should be ok?

I'm using this init method:
`tft.init(240, 240);           // Init ST7789 240x240`

I've also tried the following as I found it on a blog post:
`tft.init(240, 240, SPI_MODE2);           // Init ST7789 240x240`

But still I only get a black screen.

I'm using the latest library manager version of This library, Adafruit GFX and Adafruit BUSIO

Anything you can suggest to help get it working?

ladyada, 2020-12-04:
yeah we dont support the unusual way they set up these displays - CS is required!  please ask the vendor for some example code to get you going :) what they did basically makes the displays unusable, we have no idea why the left off the most important pin :(

closed 2020-12-04 by ladyada

ozett, 2020-12-20:
https://www.instructables.com/Adding-CS-Pin-to-13-LCD/