adafruit/Adafruit-ST7735-Library #137: offset issue with 1.14" 240x135 color TFT

centimeter, 2021-03-25:
- Arduino board:  Arduino NANO IOT 33

- Arduino IDE version (found in Arduino -> About Arduino menu):  1.8.13

Hi, as of the newest update (1.7.0) the library offsets everything on the 1.14" TFTs. (image displaces near the center of the screen, offset by static on the left and top) Looking at the library, I think this might be due to commit 2de13766b51a2b285d18150ce4903ae7dc0f9a9d, which added another if statement for another display size, but now may incorrectly set the offset for the 1.14" screens (which are 240x135)?

I'm sadly a meche student and don't have other displays to test/am not confident/skilled enough to like, actually change the code and am just rolling back for the time being but it seems like a potentially easy fix? Apologies if this is a misuse of an issue haha.

makermelissa, 2021-03-25:
I see. It should probably be an `else if` statement instead of separate if and if else statements. Want to submit a pull request?

makermelissa, 2021-03-25:
Actually, it's quick, so I'll do it.

PR #138 (2021-03-25): Fix else/if statement for 1.14-inch display
Fixes #137.
closed 2021-03-25 by makermelissa