adafruit/Adafruit-ST7735-Library #139: Strange offset when using a ST7789V

T94T, 2021-04-01:
We got a ST7789V display to work reasonably well with this library.
But in use with `tft.init(135, 240);` there is an offset:
[image]

For example, if we use `tft.init(240, 280);` we get this result:
[image]

makermelissa, 2021-04-01:
Hmm, is that a different brand? If so, it looks like they are using different offsets than the Adafruit ones even with the same screen size, so we may have to come up with a slightly different solution than just looking at size.

ladyada, 2021-04-01:
@T94T please submit a PR for your display when you get it working!

closed 2021-04-01 by ladyada