adafruit/Adafruit-ST7735-Library #147: How to use a RGB color ?

Gruniek, 2021-07-02:
- Arduino board:  ESP32
- Arduino IDE version :  1.8.15

Hi,

My question is simple, how to set a custom color like RGB code ? 

In you're library, it's a HEX data and I would like to use like : (255,255,0)
How I can do ? It's possible to convert the RGB values in you HEX value ?

// Some ready-made 16-bit ('565') color settings:
#define ST77XX_BLACK **0x0000**
#define ST77XX_WHITE **0xFFFF**
#define ST77XX_RED **0xF800**
#define ST77XX_GREEN **0x07E0**
#define ST77XX_BLUE **0x001F**
#define ST77XX_CYAN **0x07FF**
#define ST77XX_MAGENTA **0xF81F**
#define ST77XX_YELLOW **0xFFE0**
#define ST77XX_ORANGE **0xFC00**

Thanks !

dherrada, 2021-07-02:
@Gruniek I think this is what you're looking for. https://www.google.com/search?q=rgb+to+hex&oq=rgb+to+hex+&aqs=chrome..69i57.2520j0j7&sourceid=chrome&ie=UTF-8

However, if you want to do it in the code, here's how you'd do that https://forum.arduino.cc/t/rgb-to-hex/114837

ladyada, 2021-07-02:
use color565 func

closed 2021-07-02 by ladyada

Gruniek, 2021-07-20:
Thanks a lot !