adafruit/Adafruit-ST7735-Library #152: method to handle tearing effect line does not send the required argument to the display driver

michal037, 2021-10-03:
According to the [documentation of the ST7735R driver](https://cdn-shop.adafruit.com/datasheets/ST7735R_V0.2.pdf), after sending the command, an argument have to be sent later _(see page 112)_. The argument contains one bit that sets how tearing effect line is to be configured. There is no code in the library to send this argument.

https://github.com/adafruit/Adafruit-ST7735-Library/blob/8a3acfeef57f2847671e402299341aa3be8bb91f/Adafruit_ST77xx.cpp#L237

```Cpp
void Adafruit_ST77xx::enableTearing(boolean enable) {
  sendCommand(enable ? ST77XX_TEON : ST77XX_TEOFF);
}
```

---

This line is physical and should be connected to the microcontroller in order to be used. There is a bug in the PDF documentation and it is missing in the Block diagram _(see page 12)_. But in the `Pad Center Coordinates` section of the document it is marked with the name `TE` _(see page 6 and pin number 81)_. None of the products below have this physical line:

```
 Works with the Adafruit 1.8" TFT Breakout w/SD card
    ----> http://www.adafruit.com/products/358
  The 1.8" TFT shield
    ----> https://www.adafruit.com/product/802
  The 1.44" TFT breakout
    ----> https://www.adafruit.com/product/2088
  as well as Adafruit raw 1.8" TFT display
    ----> http://www.adafruit.com/products/618
```

ladyada, 2021-10-03:
ok we did test it successfully a while ago, our 1.54" breakout has a TE line - you could submit a tested PR if ya like

michal037, 2021-10-03:
I did not check other drivers which this library supports. I'll do it soon. I can send a PR, but I'd prefer to discuss how to design the code first.

enabling requires the flag, disabling does not require a flag;

What do you think of the two methods? One for enabling with a flag as a parameter. One to disable without flag. I am asking because this code is for an Arduino which has its own style. Another problem is a change that won't break people's programs.

I'm thinking of something like that, it should be safe:

```C
Adafruit_ST77xx::enableTearing(boolean mode);
Adafruit_ST77xx::disableTearing()
```

ladyada, 2021-10-03:
best to keep the same function name, can add another argument.