adafruit/Adafruit-ST7735-Library #154: Ability to change BGR / RGB method (for some AliExpress INITR_GREENTAB models)

simkard69, 2021-11-01:
Hello,

I bought the following screen on AliExpress (but can be found anywhere nowadays) :
[image]

Seems like the RGB coding that this screen expects is **ST77XX_MADCTL_RGB** instead of **ST7735_MADCTL_BGR**
Initializing it with BLACKTAB works for color but gets bottom and right edges wrong.
Initializing it with GREENTAB works for edges but not for colors

Seems like people are referring to these screens as YELLOWTAB ... anyway these ones seem to be a little bit annoying to use.

In order to solve out the problem I had to modify the library file "Adafruit_ST7735.cpp" and change these lines :

...
  switch (rotation) {
  case 0:
    if ((tabcolor == INITR_BLACKTAB) || (tabcolor == INITR_MINI160x80)) {
      madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB;
    } else {
      // madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST7735_MADCTL_BGR;
	  **madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB;**
    }
...
  case 1:
    if ((tabcolor == INITR_BLACKTAB) || (tabcolor == INITR_MINI160x80)) {
      madctl = ST77XX_MADCTL_MY | ST77XX_MADCTL_MV | ST77XX_MADCTL_RGB;
    } else {
      // madctl = ST77XX_MADCTL_MY | ST77XX_MADCTL_MV | ST7735_MADCTL_BGR;
	  **madctl = ST77XX_MADCTL_MY | ST77XX_MADCTL_MV | ST77XX_MADCTL_RGB;**
    }
...
  case 2:
    if ((tabcolor == INITR_BLACKTAB) || (tabcolor == INITR_MINI160x80)) {
      madctl = ST77XX_MADCTL_RGB;
    } else {
      // madctl = ST7735_MADCTL_BGR;
	  **madctl = ST77XX_MADCTL_RGB;**
    }
...
  case 3:
    if ((tabcolor == INITR_BLACKTAB) || (tabcolor == INITR_MINI160x80)) {
      madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MV | ST77XX_MADCTL_RGB;
    } else {
      // madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MV | ST7735_MADCTL_BGR;
	  **madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MV | ST77XX_MADCTL_RGB;**
...
    
[Adafruit_ST7735.zip](https://github.com/adafruit/Adafruit-ST7735-Library/files/7452332/Adafruit_ST7735.zip)

theloukou, 2022-03-21:
Hey, please try out my forked library version, it should be working with your screen (use init INITR_GREENTAB_WS)
If so, leave a note on PR #168 to confirm it works correctly.

IAmOrion, 2022-10-25:
+1 - I have this exact same issue and came across this issue searching a way to change the RGB/BGR method at runtime rather than having to edit the libraries.  Would be nice for a GREENTAB with RGB as an option (So GREENTAB col/row stuff with BLACKTAB colour profile)

theloukou, 2022-10-25:
Hey @IAmOrion, please try my forked version if you haven't already.
If everything is OK to you as well, post a comment or something in PR #168, maybe someday someone will see them :P

IAmOrion, 2022-10-25:
> Hey @IAmOrion, please try my forked version if you haven't already. If everything is OK to you as well, post a comment or something in PR #168, maybe someday someone will see them :P

I got mine working by just modding the library to be RGB with green tab instead BGR but yes, it would be nice if they could pull in the fixes