adafruit/Adafruit-ST7735-Library #157: ST7735 lib  -  Text mirror problem

classic-audio, 2021-11-17:
Display:  80x160 color TFT display (driver ST7735S SPI ) 
Arduino:  Arduino ProMini 3V3

I have downloaded the latest Adafruit-ST7735-Library yesterday.

The problem:
The text is displayed mirrowed (see pickture below)

The question: how can I correct that problem?

 
[image]

The test sketch:

#include <Adafruit_GFX.h>
#include <Adafruit_ST7735.h>

Adafruit_ST7735 lcd = Adafruit_ST7735(9,10,8);
//--- LCD color def. --------------------------------                                
#define BLACK 0x000                   
#define BLUE 0x001F                                       
#define GREEN 0x07E0                                   
#define YELLOW 0xFFE0            
#define WHITE 0xFFFF
#define GREY 0x652F
#define RED 0xF800               
#define OLIVE 0x7BE0  
#define ORANGE 0xFD20

void setup(void){
// --- Display -----------------------------------
  lcd.initR(INITR_MINI160x80);

  lcd.invertDisplay(1);
  lcd.setRotation(2);
  lcd.fillScreen(BLACK);
}

void loop(){
    lcd.setTextColor(GREEN);
  lcd.setCursor(0,0);lcd.print("CLASSIC AUDIO");
    lcd.setTextColor(RED);
  lcd.setCursor(0,15);lcd.print("DESIGN");
    lcd.setTextColor(YELLOW);
  lcd.setCursor(0,25);lcd.print("Attenuator");
while(1);
}

davidneosys, 2022-11-03:
hi,
did you solve that problem? I also had same problem. please share your modification.

ideva-SAS, 2022-12-02:
hi

change ST77XX_MADCTL_MY 0x80 to 0x00 in Adafruit_ST77xx.h

YonLiud, 2024-06-30:
> change ST77XX_MADCTL_MY 0x80 to 0x00 in Adafruit_ST77xx.h

I got a different fix, maybe it's my display but your approach was right, but instead of flipping the image to the Y Axis, I needed to flip to the X axis, I assume classic-audio did need that as well.

<br />

```diff
#define ST77XX_MADCTL_MY 0x80
- #define ST77XX_MADCTL_MX 0x40
+ #define ST77X_MADCTL_MX 0x00
#define ST77XX_MADCTL_MV 0x20
#define ST77XX_MADCTL_ML 0x10
#define ST77XX_MADCTL_RGB 0x00
```