adafruit/Adafruit-ST7735-Library #164: not spout esp32c3 HW SPI?

wx85105157, 2022-01-20:
- Arduino board:  **ESP-C3-13-Kit**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.19**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **LIST REPRO STEPS BELOW**

wx85105157, 2022-01-23:
After line 582 of the Adafruit_SPITFT.cpp file:
   if (!freq)
     freq = DEFAULT_SPI_FREQ; // If no freq specified, use default
Add the following code:
freq = 10000000;

After line 639:
  hwspi._spi->begin();
Add the following code:
       pinMode(3, OUTPUT);//spi clk
       pinMode(4, OUTPUT);//spi mosi
       gpio_matrix_out(3, 63, false, false);//spi clk
       gpio_matrix_out(4, 65, false, false);//spi mosi
       gpio_matrix_out(5, 0x100, false, false);
       gpio_matrix_out(6, 0x100, false, false);
       pinMode(5, OUTPUT);
       pinMode(6, OUTPUT);

The LCD works perfectly on the hardware spi.

ladyada, 2022-05-08:
can you try again with the 2.0.3 ESP for arduino ide release?