adafruit/Adafruit-ST7735-Library #170: Arduino Nano Hardware SPI support broken

username7291, 2022-05-17:
Arduino Nano 16MHZ, 5V  1.8" TFT Display 7735 128x160 with onboard Levelshifter and 3V3 Regulator. SPI HW support is broken with this library. Only slow mode can be used. 
I found with google people have similar problems but no solution. Cause this lib is very big and passes functions around i cant fix it myself.

#define TFT_CS        -1 // its to ground only 1 device
#define TFT_RST       1 // Connecting to Arduino RST dont work
#define TFT_DC         12  //   RS Data/Command
#define TFT_MOSI 11  // Data out
#define TFT_SCLK 13  // Clock 

Works but slow without Hardware SPI, SPI Settings has no affect
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST); 

Dont work, display stays white(too with TFT_MOSI/SCLK define removed):
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);

ladyada, 2022-05-17:
what if you use TFT_CS?

username7291, 2022-05-17:
In the sourcecode of the library i read -1 is supported. I set it to 14 or 53 dont work too. I think in a test i had original Pin numbers including CS connected and had the same problem. All Digital and Analog Pins are in use even 0/1.

ladyada, 2022-05-18:
are you using any hardware purchased from adafruit?

username7291, 2022-05-19:
Yes. With original Pin numbers i mean the default numbers in the source. Cause i need all possible Arduino pins i grounded cs later. Cause of space requirements SD port is removed.

ladyada, 2022-05-21:
we know 32u4 works, so not sure what to say, maybe you have bad hardware

username7291, 2022-05-23:
2 Displays and 2 Arduinos? Maybe its cause there is something on SPI SS/MOSI Pin, dont know. I optimized the library its now about 4 times fast with software SPI. Cant compare to Hardware SPI, a bit slower than 2.4" parallel display on Arduino Pro Mini 8 MHZ.

ngoan98tv, 2023-08-06:
Hi guys, I just faced the same issue.

Default SPI not works
```cpp
#define TFT_CS 10
#define TFT_RST 9
#define TFT_DC 8

Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);
```

But this works
```cpp
#define TFT_CS 10
#define TFT_RST 9
#define TFT_DC 8

#define TFT_MOSI 11 // Data out
#define TFT_SCLK 13 // Clock out

Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);
```

I'm not sure about the speed, but it seem an issue.

chupocro, 2024-04-26:
Same here, with Mega 2560 and hardware SPI the screen is white but software communication does work while with Uno hardware SPI works well.