adafruit/Adafruit-ST7735-Library #180: Blue color displayed as RED

brightproject, 2023-03-07:
#define BLUE 0x001F /* 0, 0, 255 */
tft.fillScreen(BLUE);
Why?

tadamx, 2023-03-23:
Hi @brightproject , 
Maybe used GREENTAB or default nothing in InitR? 
Try tft.initR(INITR_BLACKTAB); in void setup section.
tadamx

brightproject, 2023-03-23:
GREENTAB  - initialisation GOOD, but image shifted.
I corrected file Adafruit_ST7735.cpp
```
//madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST7735_MADCTL_BGR;
madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB;
```

samo-sk, 2023-08-11:
> GREENTAB - initialisation GOOD, but image shifted. 

Did you mean BLACKTAB? Because your corrected version of `Adafruit_ST7735.cpp` would cause BLACKTAB colors to be used. It would also cause the image to be shifted if it is the same display type as in #191 and #154.