adafruit/Adafruit-ST7735-Library #191: Deek-Robot TFT

samo-sk, 2023-08-07:
Hello,
 I have a "Deek-Robot" TFT, which looks exactly like one pictured [here](https://www.aliexpress.com/item/32855569797.html), and uses ST7735. None of the options for `initR` works well – the screen size is OK for `INITR_GREENTAB` and `INITR_BLACKTAB`, but when using `INITR_GREENTAB`, red is swapped with blue, and when using `INITR_BLACKTAB`, the image on the display is shifted. This issue is the same as the one described in #180, but I'm not sure what type of display is used by the user who reported it. This issue is also described [here](https://forum.arduino.cc/t/deek-robot-shield/523051). 
The library works perfectly when `initR(INITR_BLACKTAB)` is used, but it is modified to set _colstart to 2 and _rowstart to 1 (like when `INITR_GREENTAB` is used). Could you please modify this library (more specifically, `Adafruit_ST7735.cpp` and `Adafruit_ST7735.h`) to create another option for `initR` for initializing the display like mentioned above? Also `setRotation` will need to be modified to behave just like when `tabcolor` is `INITR_BLACKTAB` when the new option is used.
Thank you.

ladyada, 2023-08-09:
please ask DEEK to submit a PR to help and we will review it!

closed 2023-08-09 by ladyada

samo-sk, 2023-08-10:
I have noticed this issue is a duplicate of #154.

samo-sk, 2023-08-11:
Information for anybody reading this issue report after its closure:
The proposed fix is (in theory) incorrect, as the values of offsets (`_colstart` and `_rowstart`) are also present in `Rcmd2green`, a part of initialization code for displays with the mentioned offsets. However, the proposed change incorrectly uses the init code for blacktab (`Rcmd2red`), which is offset-less. However, this issue has no effect in practice, as the effects of `Rcmd2green`/`Rcmd2red` are overriden by `Adafruit_ST77xx::setAddrWindow`, which is called before anything is drawn on the display.
PS: You may be wondering what are the `Rcmd2[...]` display commands for, as they effects are overriden by drawing functions. I don't know the answer. However this may be related to the fact that there is a deprecated function, `pushColor`, in [Adafruit_SPITFT.cpp](https://github.com/adafruit/Adafruit-GFX-Library/blob/master/Adafruit_SPITFT.cpp), which doesn't call `setAddrWindow`.