adafruit/Adafruit-ST7735-Library #197: ST7789(GMT130v1) does not work with Wemos D1 mini

3ton, 2024-04-29:
GMT130v1                         Wemos D1 mini

     GND                                  GND (15)
     VCC                                   3V3 (8)
     SCK                                    SCK/D5  (5)
     SDA                                   MOSI/D7 (7)
     RES                                    D4 (17)
     DC                                     D3 (18)
     BLK                                    -----

[image]

[image]

3ton, 2024-04-29:
Ah, sorry.
SPI_MODE3 resolve my problem

gkostov, 2025-02-21:
Thanks a lot for the hint!! I've had this built and running with SPI_MODE2 some years ago and after trying to re-flash it recently it simply did not work! Yes, SPI_MODE3 is what it takes to get it working.

Cheers