adafruit/Adafruit-ST7735-Library #204: st7789 without cs pin

JackIsGoofingOff, 2024-10-27:
Hi, sorry to border you guys, I am running into an issue when controlling a st7789 without **CS** pin.  The board I am using is an Esp-32 module and here is the config: 
`
// Pin configs
#define TFT_CS    -1
#define TFT_RST   13
#define TFT_DC    2
#define TFT_SCLK  18
#define TFT_MOSI  23

// Display initialization
//Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_RST);
Adafruit_ST7789 tft = Adafruit_ST7789(TFT_CS, TFT_DC, TFT_MOSI, TFT_SCLK, TFT_RST);
`
I also tried to add SPI_MODE3 in tft.init() function but still noting came out from the screen. Are there any solutions to drive this screen without CS pin? Thank you so much.

caternuson, 2024-10-27:
The CS pin can not be ignored when using SPI in general. The CS pin of the ST7789 needs to be properly handled somehow. How are you currently asserting CS for the ST7789?

JackIsGoofingOff, 2024-10-28:
@caternuson Thanks. The thing is that my ST7789 does not contain a CS pin (or somehow grounded). With TFT_eSPI library from Bodmer things can work by setting -1 to CS, however this cannot work in Adafruit library. And for some cases I want to run examples from Adafruit library hence have the above question.

JazzeaN, 2024-11-08:
Hello.
This works for me,  ESP32 and ST7789 240x240 1.3 inch TFT  display.

`  #include <Adafruit_GFX.h>    // Core graphics library
  #include <Adafruit_ST7789.h> // Hardware-specific library for ST7789
  #include <SPI.h>

  // ESP32 
  #define TFT_RST        15
  #define TFT_DC         32

  Adafruit_ST7789 tft = Adafruit_ST7789(-1, TFT_DC, TFT_RST);

void setup(void) {
  Serial.begin(9600);
  Serial.print(F("Hello! ST7789 TFT Test"));

  //   Init ST7789 240x240 1.3" or 1.54"  TFT:
  tft.init(240, 240, SPI_MODE2);    
         
  Serial.println(F("Initialized"));
  tft.setRotation(3);
  tft.fillScreen(ST77XX_BLACK);
  tft.setTextWrap(false);
  tft.setCursor(0, 30);
  tft.setTextColor(ST77XX_RED);
  tft.setTextSize(1);
  tft.println("Hello World!");
  tft.setTextColor(ST77XX_YELLOW);
  tft.setTextSize(2);
  tft.println("Hello World!");
  tft.setTextColor(ST77XX_GREEN);
  tft.setTextSize(3);
  tft.println("Hello World!");
  tft.setTextColor(ST77XX_BLUE);

}

 void loop() {
}`

Sxs7513, 2024-11-09:
@JackIsGoofingOff have u got any solutions？

gkostov, 2025-02-21:
I've been using it with SPI_MODE2 some years ago but weirdly these days it takes SPI_MODE3:
```
tft.init(240, 240, SPI_MODE3);
```
See https://github.com/adafruit/Adafruit-ST7735-Library/issues/197#issuecomment-2083200043