adafruit/Adafruit-ST7735-Library #208: Flicker issue is not smooth when blinking

BimoSora2, 2025-01-23:
I hope you fix the library to overcome flicker like this which is clearly not smooth.

https://github.com/user-attachments/assets/d29f143c-47bc-4c2c-843e-31013ca48fa7
```
#include <Adafruit_GFX.h>
#include <Adafruit_ST7735.h>
#include <Adafruit_BMP280.h>
#include <Adafruit_MPU6050.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_INA219.h>
#include <SPI.h>
#include <Wire.h>
#include <LittleFS.h>
#include <TimeLib.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <DNSServer.h>
#include <ArduinoJson.h>
#include <Fonts/Swansea5pt7b.h>
#include <Fonts/Swansea6pt7b.h>
#include <Fonts/Open_24_Display_St17pt7b.h>

const char* ssid = "Time";         
const char* password = "12345678";    

#define TFT_MOSI  13
#define TFT_SCLK  14
#define TFT_DC    0
#define downButton 12

const int BATT_IMG_WIDTH = 22;
const int BATT_IMG_HEIGHT = 10;
const int ICON_WIDTH = 80;
const int ICON_HEIGHT = 80;

const int SMALL_ICON_WIDTH = 32;
const int SMALL_ICON_HEIGHT = 32;

const char* ICON_TEMP = "/temp_icon.bmp";
const char* ICON_PRESSURE = "/pressure_icon.bmp";
const char* ICON_ALTITUDE = "/altitude_icon.bmp";
const char* ICON_ANGLE = "/angle_icon.bmp";
const char* ICON_STEP = "/step_icon.bmp";
const char* ICON_SETTINGS = "/settings_icon.bmp";

const byte DNS_PORT = 53;
DNSServer dnsServer;
IPAddress apIP(172, 217, 28, 1);

ESP8266WebServer server(80);
Adafruit_ST7735 tft = Adafruit_ST7735(-1, TFT_DC, TFT_MOSI, TFT_SCLK, -1);
Adafruit_BMP280 bmp;
Adafruit_MPU6050 mpu;
Adafruit_INA219 ina219;

bool bmpEnabled = false;
bool mpuEnabled = false;

const float BATTERY_MAX = 4.2;
const float BATTERY_MIN = 3.0;

float temperature = 0, pressure = 0, altitude = 0;
float lastPressure = 0;
String tempC = "0.0 *C", tempF = "0.0 *F", pressureStr = "0.0 hPa", altitudeStr = "0.0 m";
String accX = "X: 0*", accY = "Y: 0*", accZ = "Z: 0*";
String batteryStatus = "100%";
String prevBatteryStatus = "";
String lastVoltageStr = "";
String prevTempC, prevTempF, prevPressureStr, prevAltitudeStr;
String prevAccX = "", prevAccY = "", prevAccZ = "";
[... 1274 lines cut ...]
            autoReturnEnabled = false;
            if (!wifiInitialized) {
                WiFi.mode(WIFI_AP);
                WiFi.softAP(ssid, password);
                setupWebServer();
                wifiInitialized = true;
                Serial.println("Access Point Started");
                Serial.print("IP Address: ");
                Serial.println(WiFi.softAPIP());
            }
            
            tft.setFont(&Swansea6pt7b);
            drawMenuIcon(ICON_SETTINGS, CONTENT_MARGIN + 25);
            
            if (settingsText != prevSettingsText) {
                int yBase = CONTENT_MARGIN + ICON_HEIGHT + 0;
                int yStep = 30;
                drawCenteredText(settingsText, yBase + yStep + 0, safeColor565(255, 255, 255));
                prevSettingsText = settingsText;
            }
        }
        break;
    }
}
```

OtterCodes101, 2025-07-14:
there's an easy way to mitigate this. just check if the icon actually needs to be updated instead of always redrawing it no matter what

OtterCodes101, 2025-07-14:
oh wait nvm i wasn't paying attention enough sorry

OtterCodes101, 2025-07-14:
checking with my edited version of the graphics test, only the initial text "lorem ipsum" text has the same effect as your icon. everything else doesn't have that.
even stranger, i actually cannot tell why this is.