adafruit/Adafruit-ST7735-Library #213: 2.8" 240x320 Display with touch and ST7789 driver has colors inverted

pdfelen, 2025-03-13:
With some help browsing the internet I did found a solution for my problem with the 2.8" 240x320 display. After initialization the colors are inverted.

In file "Adafruit_ST7789.cpp" on line 72 the command structure 'generic_st7789' holds setting ST77XX_INVON, for my screen this needs to be ST77XX_INVOFF.
After changing that my screen works. 
For now I duplicated the Adafruit_ST7789 .cpp and .h files into Adafruit_ST7789_28.cpp and .h and made the change. A more permanent solution would be appreciated where this is user configurable.

Thanks

Voss-Leonard, 2025-03-14:
@pdfelen 

I have exactly the same problem. 
This setting must also be changed for the following displays for electronics housings from Phoenix Contact! 

1. https://www.phoenixcontact.com/en-us/products/housing-cover-bc-1076-dkl-r-d24-kpg-kmgy-1335860
2. https://www.phoenixcontact.com/en-us/products/component-housing-upper-part-ics50-c100x12-d24-trg-7035-1077656

cloud6696, 2026-08-17:
I encountered the same problem. I didn't change anything in the libraries, I just added a line of code after the display initialization.
tft.invertDisplay(false);